fr3ts0n/AndrOBD #85: Crash on attempt to connect to paired device

paulcor, 2019-09-20:
Steps to reproduce the behavior:
Click plug icon, click on paired device listed below "Select adapter", --> "AndrOBD has stopped".
```
09-20 09:07:13.331 28548 28548 E AndroidRuntime: FATAL EXCEPTION: main
09-20 09:07:13.331 28548 28548 E AndroidRuntime: Process: com.fr3ts0n.ecu.gui.androbd, PID: 28548
09-20 09:07:13.331 28548 28548 E AndroidRuntime: java.lang.RuntimeException: Failure delivering result ResultInfo{who=null, request=2, result=-1, data=Intent { (has extras) }} to activity {com.fr3ts0n.ecu.gui.androbd/com.fr3ts0n.ecu.gui.androbd.MainActivity}: java.lang.NullPointerException: Attempt to get length of null array
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at android.app.ActivityThread.deliverResults(ActivityThread.java:4268)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at android.app.ActivityThread.handleSendResult(ActivityThread.java:4312)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at android.app.ActivityThread.-wrap19(Unknown Source:0)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1644)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:106)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:164)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6494)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to get length of null array
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at com.fr3ts0n.ecu.gui.androbd.BtCommService$BtConnectThread.logSocketUuids(BtCommService.java:257)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at com.fr3ts0n.ecu.gui.androbd.BtCommService$BtConnectThread.<init>(BtCommService.java:241)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at com.fr3ts0n.ecu.gui.androbd.BtCommService.connect(BtCommService.java:131)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at com.fr3ts0n.ecu.gui.androbd.MainActivity.connectBtDevice(MainActivity.java:1702)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at com.fr3ts0n.ecu.gui.androbd.MainActivity.onActivityResult(MainActivity.java:685)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at android.app.Activity.dispatchActivityResult(Activity.java:7276)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	at android.app.ActivityThread.deliverResults(ActivityThread.java:4264)
09-20 09:07:13.331 28548 28548 E AndroidRuntime: 	... 9 more
```
**AndrOBD Debug log files**
```
2019-09-20	09:32:42.480	INFO	AndrOBD	AndrOBD V2.0.7 starting
2019-09-20	09:32:42.481	FINER	com.fr3ts0n.pvs	+PvListener:com.fr3ts0n.pvs.PvList[null]->com.fr3ts0n.ecu.gui.androbd.MainActivity@b87d27d
2019-09-20	09:32:42.481	FINER	com.fr3ts0n.pvs	+PvListener:com.fr3ts0n.pvs.PvList[null]->com.fr3ts0n.ecu.gui.androbd.MainActivity@b87d27d
2019-09-20	09:32:42.482	FINER	com.fr3ts0n.pvs	+PvListener:com.fr3ts0n.pvs.PvList[null]->com.fr3ts0n.ecu.gui.androbd.MainActivity@b87d27d
2019-09-20	09:32:42.489	FINE	AndrOBD	Adapter: android.bluetooth.BluetoothAdapter@f713a9a
2019-09-20	09:32:42.574	INFO	AndrOBD	Conversion: metric
2019-09-20	09:32:47.331	FINE	BtDeviceListActivity	Sending Result...
2019-09-20	09:32:47.398	FINE	CommService	connect to: AA:BB:CC:11:22:33
2019-09-20	09:32:47.399	FINE	CommService	setState() NONE -> CONNECTING
```

**Important details on your environment:**
- Android OS version - 8.1.0
- OBD adapter type - generic bluetooth dongle

paulcor, 2019-09-20:
Looking at the source code, the crash is in code for logging at INFO level. Setting the log level to WARNING avoids the problem, and I am able to connect to the ECU and read DTCs.

fr3ts0n, 2019-09-21:
Hi @paulcor 
Thanks for the input.

It appears that your adapter has a very poor implementationof bluetooth.
It obviously does not report any UUID which, by standard, identifies the adapter's purpose / features.

However, this will be fixed in AndrOBD, since it should definitely not crash the app.

commit d8638bb2 (2019-09-21): Fix #85: Robust handling of BT socket w/o reported UUIDs
closed 2019-09-21 by fr3ts0n