fr3ts0n/AndrOBD #89: other ECUs ?

freeeflyer, 2019-11-11:
Hello,
Is there a way to access other ECU like mrs/srs or others ?
thanks

fr3ts0n, 2019-11-11:
No, usually only emissions relevant ECUs of the drive train support the OBD protocol.

freeeflyer, 2019-11-12:
Are you sure ? I accessed airbag and all other ECU with an android commercial closed source solution and my ELM327..

ChrisS81, 2019-11-12:
My Car has 3 ECUs on a CAN Bus and i can connect to any of them using AndrOBD without issues. 
Sure, some if them are sending Nonsens data for some Codepoints but nonetheless i can connect to them.
The first one AndrOBD shows me is the Engine ECU. The second one is Transmission Control and the third is something like exhaust Processing IIRC (AdBlue system i guess)

fr3ts0n, 2019-11-12:
> Are you sure ? I accessed airbag and all other ECU with an android commercial closed source solution and my ELM327..

Other ECUs may be accessed using different communication protocol(s) than OBD. 
These protocols  are only partially standardized and their data content is very specific to the vehicle manufacturer- and even vehicle type.

That's why these apps usually are closed source, specific to certain manufacturers and commercial.
Some of these even can operate on ELM327 adapters.

fr3ts0n, 2019-11-12:
@ChrisS81 I agree, all three ECU's you can connect with AndrOBD are emissions relevant and obviously service OBD protocol. 
However, chances are poor to find an Airbag- or SRS-ECU which serves OBD protocol.

enfranc1991, 2020-10-06:
555