fr3ts0n/AndrOBD #90: Can't fetch OBD Data, Service not (yet) supported: 128

landoulsi, 2019-12-29:
**Describe the bug**
I am not able to connect to the obd simulator except very few times, it's like I can connect to the OBD simulator only 10% of times, in rest of cases I am getting this error:
```
'[4, 1, 0, 0, B, E, 1, B, 3, 0, 1, 3, 0, 0, 8, 8, 1, 8, 0, 0, 1, 0, B, C, 4, 8]':For input string: "BE1B30130088180010BC48"
Service not (yet) supported: 128
```

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to the OBD
2. Select ECU (if select ECU dialog appears)
3. Press on the menu arrow
4. Select `OBD Data`
5. Nothing appear on the screen and on the logs I can see the error above

**Expected behavior**
See OBD Data, like speed, acceleration ...

**AndrOBD Debug log files**
[data_connection_fail_logs.log](https://github.com/fr3ts0n/AndrOBD/files/4008009/data_connection_fail_logs.log)

AndrOBD stores log files in a subdirectory on your primary storage device. (com.fr3ts0n.ecu.gui.androbd/log/AndrOBD.log.*)

**Log detail level**
The detail level of the logged debug information can be adjusted in the Settings > Development options > Logging level.

By default the detail level is set to INFO.
Preferred detail level for issue analysis is DEBUG. (Since V1.5.0 this level is called FINE)

**Important details on your environment:**
- Android OS version: Android Oreo
- AndrOBD version: Latest from master branch
- OBD adapter type: Bluetooth
- Connected vehicle make, type, year, engine size: OBD simulator ECU SIM 2000 Professional ISO 9141-2

fr3ts0n, 2020-01-11:
Thanks for your report and the proposed changes in pull request #91 .

I'm not sure if this is a real world situation coming from a OBD simulator, since OBD defines it's PID support bits to be limited to 32-Bit blocks.

However, let's do the changes to get it robust :+1: 

Please see my comments in pull request #91

fr3ts0n, 2020-01-31:
Unfortunately we could not get your PR #91 cleaned up.
Changes are now implemented from my side with commit 67e71c936673aede40b0213e3d0f3f9dcc1a34ac.

closed 2020-01-31 by fr3ts0n