fr3ts0n/AndrOBD #95: Check background picture is distracting

aha999, 2020-01-15:
In the Obd Data menu, the Check Engine wallpaper/background is distracing so the text is not as visible.

I recommend that the icon is moved to the top bar, and the background is changed.

fr3ts0n, 2020-01-17:
The big check engine light is designed that way.
It shall be easily visible in case OBD protocol indicates the check engine light to be ON.
- In full screen mode
- With status bar disabled

closed 2020-01-17 by fr3ts0n

aha999, 2021-05-05:
Maybe you could use a first row for displaying a text: **MIL Status: ON/OFF** and a small version of this logo that indicates it's on (bright yellow) or off (dark), like in a car. This way it would look much better. 
i believe not a lot of people would assume that the background could mean that the MIL is actually on, because if it is, it should be displayed like other info.

blommetjie, 2021-05-06:
Nothing coudent do noting to mil

fr3ts0n, 2021-05-07:
I really think it is fine to display an active MIL like it currently is. 
If you regularly test and measure your vehicle, it does not affect you at all, but if your MIL is active it shall be distracting somehow, since it is supposed to warn you of a problem with your car.

aha999, 2021-05-07:
yes, but the point is that MIL is massive and in the background so it is distracting. On the top it says "MIL status ON" anyways... maybe you can add a small MIL logo to that line... And the background would look cleaner