fr3ts0n/AndrOBD #97: Long pressing the DTC doesn't open a website / info

aha999, 2020-01-15:
When long pressing a DTC code, it doesn't open anthying like details, but throws an error. I think it should open the default web browser for example / trouble code lookup

[image]

fr3ts0n, 2020-01-17:
What kind of browser do you have installed on your device?
This is a standard Android web search request which triggers the browser to open a search for "OBD <DFC#>"
I haven't heard of any other reports of this not working ...

aha999, 2020-01-18:
Maybe it's indeed a problem on my side because i've had this problem before recently on another program...

aha999, 2020-02-12:
The problem was on my side, the browser had a bug that got fixed. Closing the issue.

closed 2020-02-12 by aha999