fr3ts0n/AndrOBD #105: Add 'not installed' plugin info to Plugin manager

aha999, 2020-01-23:
You should add 'not installed' plugins to plugin manager, so users know what plugins exist and can download them from there, or have links there...

fr3ts0n, 2020-01-24:
Since the list of available plugins will be growing, I don't think it should be maintained into AndrOBD itself, but I agree that a (F-Droid-) search for available plugins is quite helpful.

aha999, 2020-02-11:
What if you added "install plugins" button into the Plugin manager, that would sync with the latest plugins that are available ? The database could be updated/managed on your plugin repo.

That way, if we clicked "install plugins", it would automatically show us names of plugins, and be links to the f-droid at the same time...