fr3ts0n/AndrOBD #109: Error While Setting up project in android studio

pratt1305, 2020-02-10:
[image]

ERROR: Unable to resolve dependency for ':androbd@debug/compileClasspath': Could not resolve project :plugin.
Show Details
Affected Modules: androbd

ERROR: Unable to resolve dependency for ':androbd@debugAndroidTest/compileClasspath': Could not resolve project :plugin.
Show Details
Affected Modules: androbd

ERROR: Unable to resolve dependency for ':androbd@debugUnitTest/compileClasspath': Could not resolve project :plugin.
Show Details
Affected Modules: androbd

ERROR: Unable to resolve dependency for ':androbd@release/compileClasspath': Could not resolve project :plugin.
Show Details
Affected Modules: androbd

ERROR: Unable to resolve dependency for ':androbd@releaseUnitTest/compileClasspath': Could not resolve project :plugin.
Show Details
Affected Modules: androbd

fr3ts0n, 2020-02-10:
Hi @pratt1305 ,
Please check FAQ in WIKI 
https://github.com/fr3ts0n/AndrOBD/wiki/Frequently-asked-questions#build-problems---could-not-resolve-project-plugin 
for solution.

pratt1305, 2020-02-12:
Thanks For Support How can i contribute in project !!!