fr3ts0n/AndrOBD #111: generate & export diagnostic report in .html format

aha999, 2020-02-13:
If AndrOBD could export the diagnostic report in .html format, the mobile devices & desktop devices could easily recognize it / keep it readable and good-enough looking with no extra software needed and if you sent it to someone, they would just need an internet browser to open it.

Example of a html report that was generated & exported with some obd diagnostic app:

html file (zipped): [Diagnostic Report 2020-01-15.zip](https://github.com/fr3ts0n/AndrOBD/files/4201880/Diagnostic.Report.2020-01-15.zip)

Html exported file, viewed on PC:
[image]

pratt1305, 2020-02-23:
Great Idea @aha999  !!

theduskin, 2021-07-19:
i just donated a small amount, to say thank you.
and yes, this would be usefull, even in plain text, csv, whatever