fr3ts0n/AndrOBD #112: Speed & RPM dashboard

aha999, 2020-02-14:
You could make a simple Speed & RPM dashboard.

It could look like this:
[image]

[image]

Recorded from this website: http://metinseylan.com/
Source of that dashboard linked to: https://github.com/MetinSeylan/Kaput

Related to #104

aha999, 2020-07-27:
check this speedometer opensource project out. it would be awesome if you could integrate it into AndrOBD.

>  Dynamic Speedometer and Gauge for Android. amazing, powerful, and multi shape ⚡ 

https://github.com/anastr/SpeedView

fr3ts0n, 2020-07-28:
This could be quite a good alternative to the rudimentary gauges in achartengine library.
I will take a look at this.
I was hoping to find a complete replacement for the achartengine (i.e. chartfx/tilefx) but I don't want to complicate the project by adding JavaFX semantics.

fr3ts0n, 2020-07-31:
Here is a sample dash using SpeedView:
[image]

Does that look reasonable?

The only downside:
- App size is about doubled, since I need to add Kotlin support

fr3ts0n, 2020-07-31:
Test version (V2.2.0) is here:
https://t.me/c/1434285001/156

[image]

anastr, 2020-08-01:
Hi @fr3ts0n 
You can use version `1.3.0`, the last version written in Java, a lot of [changes](https://github.com/anastr/SpeedView/releases) since.

You may need to add some `sv_cutPadding` to see first and last tick.

Good work!.

fr3ts0n, 2020-08-01:
Hi @anastr 

Thanks for the hint with ```sv_cutPadding``` :+1:  

[image]

In the last java-based V1.3.0 the AwesomeSpeedometer has quite some tick scaling issues, so I'd rather have the kotlin support.

closed 2020-08-01 by fr3ts0n

aha999, 2020-08-02:
@anastr for updates/chat about this project, please join a [telegram group](https://t.me/joinchat/G60ltQv5CCEQ94BZ5yWQbg)

PR #121 (2020-08-01): Implement SpeedView Gauges
Fix #112: Integration of SpeedView gauges for nice and shiny dashboard view 
- Add kotlin support since SpeedView uses Kotlin
- Integrate SpeedView lib
- Adapt gauge view & -adapters