fr3ts0n/AndrOBD #114: Tested the app on Hyundai i10 Model - Shows everything zero after connection

pratt1305, 2020-02-20:
i tested this app with my car hyndai i10 india the devices gets connected but then it shows nothing on any of the chart. where am i going wrong ?

[image]
[image]
[image]
[image]
[image]

pratt1305, 2020-02-20:
Are there any sepecific guidelines that i have missed ?

fr3ts0n, 2020-02-20:
Hi @pratt1305 
I've never seen that one before ...
Please could you attach some log files as mentioned in https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues
Thx

pratt1305, 2020-02-21:
Log Files  As mentioned

[AndrOBD.log.3.txt](https://github.com/fr3ts0n/AndrOBD/files/4234282/AndrOBD.log.3.txt)
[AndrOBD.log.4.txt](https://github.com/fr3ts0n/AndrOBD/files/4234283/AndrOBD.log.4.txt)
[AndrOBD.log.1.txt](https://github.com/fr3ts0n/AndrOBD/files/4234284/AndrOBD.log.1.txt)
[AndrOBD.log.2.txt](https://github.com/fr3ts0n/AndrOBD/files/4234285/AndrOBD.log.2.txt)

fr3ts0n, 2020-02-22:
Thank you, you found a good one ...
I don't know what I have tested for the last release, but it was definitely not the latest build ... 

Fixed with latest commit

Test version V2.1.3 is available on [AndrOBD_DEV channel](https://t.me/AndrOBD_dev)

pratt1305, 2020-02-23:
Thanks @fr3ts0n  Issue has been solved now device is working fine i tested on 2 cars

commit 8e9f5e7f (2020-02-22): fix #114 - Correction of max payload size
closed 2020-02-23 by pratt1305