fr3ts0n/AndrOBD #117: GPS provider plugin has incorrect name/description on F-Droid

ramonsmits, 2020-05-03:
**Describe the bug**
GPS provider plugin has incorrect name on F-Droid

[image]

The description of the GPS package is identical to the MQTT package.

```txt
AndrOBD MQTTPublisher
=====================

Plugin extension for AndrOBD to allow publication of OBD measurements to a MQTT home automation server.

Functionality
-------------

- The AndrOBD MQTT publisher published OBD vehicle data gathered from AndrOBD to a configurable MQTT server.
- The extension connects to a single MQTT message broker as a publishing client using existing network connection.
- All AndrOBD data items which are selected for display/update can be selected to be published to MQTT.
- Publications are sent as a bulk of separate MQTT messages.
- Publications are triggered automatically in a cyclic loop.
- The time between publications is configurable.
- The publication may also be triggered manually fromout the host application.

Configuration
-------------

Following generic parameters for publication shall be configurable:

- MQTT parameters
- Host Name / IP address
- Port number
- User Name
- Password
- MQTT message prefix

- Update parameters
- Update cycle time [s]
- OBD data items
- Selection of subset of AndrOBD data items for display.
```

See https://f-droid.org/en/packages/com.fr3ts0n.androbd.plugin.mqtt/

**Expected behavior**

Correct information on F-Droid for the GPS package.

meyerd, 2020-06-10:
For me the sensorplugin is shown twice but with different logos.

fr3ts0n, 2020-06-10:
Yes, it seems to be a matter of the attached F-Droid package meta data.
We try to find out why it appears that way on F-Droid together with the F-Droid support guys.
For now, until this is solved, please select the plugin by icon (rather than by description) which appears to match correctly.