fr3ts0n/AndrOBD #119: Transmission Fluid Temperature?

mikegilchrist, 2020-06-13:
I've got a 2013 Toyota Sienna and am trying to check my ATF temperature per the instructions for correctly filling the transmission fluid which is supposed to be in the 100-130F range.  I can't seem to find any reading corresponding to the fluid.  What am I missing?

fr3ts0n, 2020-06-13:
Most likely ATF temperature is not supported by the OBD implementation of your car.

mikegilchrist, 2020-06-14:
Hmmm, but I see other non-open source apps offering this info, usually with a toyota specific plug in.

mikegilchrist, 2020-06-15:
Any idea what the CAT temperature readings are? There are 4 of them.

fr3ts0n, 2020-06-15:
Catalytic converter temperature.
If your car has a 2-Bank system 4 readings domake sense ...
Bank 1/2, Pre-/Post Converter

mikegilchrist, 2020-06-15:
Ahh.  That makes more sense.  

Before I close this issue, I wanted to mentionI found a way to monitor the temperature by jumping the OBD pins. I describe it [here](https://www.siennachat.com/threads/transmission-fluid-change-drain-when-oil-cold-or-warm-and-engine-running-or-stopped.63868/post-352833) (in case any one is interested)

closed 2020-06-15 by mikegilchrist

fr3ts0n, 2020-06-15:
Thanks for sharing this information