fr3ts0n/AndrOBD #120: OBDLink SX

Jastreb07, 2020-07-10:
Hi,
i cant read data with my OBDLink SX (baudrate: 115200).
it say it ist connect but it not display datat.

fr3ts0n, 2020-07-10:
Hi @Jastreb07 
Thank you for your report.

In order to answer your question I'd need some more information.

Please could you provide a log file of your connection attempt?
(see: https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues#report-issues)

Thx

Jastreb07, 2020-07-11:
[log.zip](https://github.com/fr3ts0n/AndrOBD/files/4906638/log.zip)

Jastreb07, 2020-07-11:
> Hi @Jastreb07
> Thank you for your report.
> 
> In order to answer your question I'd need some more information.
> 
> Please could you provide a log file of your connection attempt?
> (see: https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues#report-issues)
> 
> Thx

sPort.setParameters(115200, 8, UsbSerialPort.STOPBITS_1, UsbSerialPort.PARITY_NONE);

Fixed

aha999, 2020-07-27:
This might not be helpful, but it works normaly with my ObdLink MX...

fr3ts0n, 2020-07-28:
Looks like it would make sense to have USB comm parameters adjustable in app settings.

commit 33b8361e (2020-07-30): Fix #120 : USB baud rate adjustable
closed 2020-07-30 by fr3ts0n