fr3ts0n/AndrOBD #126: Main screen - Dashboard with big buttons [feature request]

aha999, 2020-08-14:
When AndrOBD starts, instead of empty screen, a convenient dashboard with big buttons (with icons) for androbd functions could appear as a home screen, which would then open a section of the app that the button represents. similar to other obd2 apps.

That would make the app look better and more convenient to use.

For example, buttons:
- Diagnostics (DTC's, Freeze frame, generating reports)
- Dashboard (speed, rpm, engine temp etc.)
- Monitors (vehicle data)
- Settings (opens a menu with settings)
- vehicle info (vin + other possible vehicle info)
- Tests (emission tests etc.)
- Connect (to the adapter, at the top or bottom) with status light icon

Other:
- AndrOBD logo at the very top (clicking it open a dialogue window, showing androbd info (about))

At the connected area, info showing:
- Adapter: Connected
- ECU: Connected

Example of the idea (hopefully a UI/UX designer steps in):
[image]

eneller, 2025-07-24:
maybe we could use a [standard navigation drawer?](https://m3.material.io/components/navigation-drawer/overview) As i see it, the mockup above would serve no functionality except as a way to get to the screen you actually want to see. Using a navigation drawer would allow for the app to launch showing the dashboard (or the last used screen) and hide away navigation for quick access

fr3ts0n, 2026-01-11:
Hi, THX for your help.
I would be glad to see a size below 10 MB. However, if practical usability really improves we should accept more.