fr3ts0n/AndrOBD #128: Follow system theme by default

aha999, 2020-09-09:
I think AndrOBD should follow system theme by default (If the system theme is dark, AndrOBD is automatically dark at startup, if it's light, AndrOBD is light). Of course there would be a manual setting, but by default is should be on 'follow system theme' in my opinion.

- Light
- Dark
- System default (the recommended default option)

More in this link:
https://developer.android.com/guide/topics/ui/look-and-feel/darktheme

fr3ts0n, 2020-09-29:
This would break backwards compatibility!
- System themes were introduced with Android 10
- AndrOBD supports Dark/Light mode (manually selected) down to Android 4.2

aha999, 2020-10-03:
Well, then if it supports it, it should follow the system theme, but if not, it would be auto-set to light mode.
Can it detect android version and hide the follow system theme if android version is older than android 10 ?
I mean, there are apps that can follow a system theme and also support older versions of android... So i don't know how that feature is implemented, nor am i a dev.

willhubs, 2021-11-25:
> Well, then if it supports it, it should follow the system theme, but if not, it would be auto-set to light mode and have a manual switch toggle in the settings.

I agree with this. this would be the ideal situation in my opinion.

RokeJulianLockhart, 2024-11-26:
As aforestated at https://github.com/fr3ts0n/AndrOBD/issues/104#issuecomment-2501200243:

> Is there a consensus that any new GUI design would be Material 3 + You? I would consider anything that doesn't adhere to it to be automatically redundant, considering how much it integrates with AOSP ≥ 12. It would remediate https://github.com/fr3ts0n/AndrOBD/issues/128#issue-697059386.