fr3ts0n/AndrOBD #130: STN chips support (ObdLink devices etc.)

aha999, 2020-10-03:
AndrOBD does currently not fully support [STN chips](https://www.obdsol.com/solutions/chips/) from Obd Solutions.

> The STN11xx/21xx OBD Interpreter ICs are based on the robust OBDLink® technology stack and provide a fast and cost-effective way to add OBD functionality to your project. All OBD interpreter ICs on this page are fully compatible with the de facto standard ELM327 command set, while outperforming the original ELM327 IC in every category: stability, performance, features and price.

I contacted ObdLink support and they provided me with a link below, to supporting their chips/adapters.

OBD Development page and STN guide:
https://www.scantool.net/forum/index.php?topic=14767.0

Included in their answer:

>You can use the FRPM listed in this page to implement support for the OBDLink line of scan tools by.

If AndrOBD would support them it would increase the obd2 adapter support, as OBDLink devices are popular and you could add "STN chips" to supported adapters/chips in the description.

richtong, 2020-11-23:
wow this would be great!

fr3ts0n, 2020-12-13:
AFAIK the STN adapters are functionally fully suported since they emulate ELM command set.
* Are there any important features missing to consider these as supported?

richtong, 2020-12-13:
Oh that’s good to know. I thought they didn’t elumate the elm instruction set. Will try that.

richtong, 2021-01-08:
Seems like you should close this issue :-)

aha999, 2021-01-09:
But can we be sure that it fully supports STN chips just by guessing ?

richtong, 2021-01-09:
I have one and can test it.

Sent via Superhuman iOS ( https://sprh.mn/?vip=rich@tongfamily.com )

aha999, 2021-01-14:
i've asked them this and this was their reply:
[image]

richtong, 2021-01-15:
> i've asked them this and this was their reply:
> slika (
> https://user-images.githubusercontent.com/50620416/104636338-7eaf2e80-56a3-11eb-9e7e-e5ce7279b84e.png
> )

Yes somethings are convenient with stn. For instance to set baud rate to highest is stbrd 2000000 whereas for commands it’s the divisor of 4000000 since the user clock runs at 4Mhz on the series bus st the command is like arbr 2

Sent via Superhuman iOS ( https://sprh.mn/?vip=rich@tongfamily.com )

richtong, 2021-01-15:
Exactly expained in https://www.scantool.net/blog/switching-communication-baud-rate/

Sent via Superhuman iOS ( https://sprh.mn/?vip=rich@tongfamily.com )

richtong, 2021-01-15:
And that should stsbr and the method described there is a little scary. It’s permanent so you can brick the device. STSBR is not. The correct soft at command at ATBRD 2

Sent via Superhuman iOS ( https://sprh.mn/?vip=rich@tongfamily.com )

aha999, 2021-05-09:
file:///C:/Users/Klemen/Downloads/Temp/1%20delete/stn1110_vs_elm327.pdf

[image]

ionality,withoutbreakingcompatibilitywiththeELM

autoMD, 2021-08-25:
ObdLink MX supported ST cmd for SWCAN (most GMC car) and MSCAN (most Ford car).
I like supper AndrOBD project. I'm reading the code, but I don't understand. I don't smarter coder but know many car protocol. :)