fr3ts0n/AndrOBD #131: Donation platforms

aha999, 2020-10-03:
**Is your feature request related to a problem? Please describe.**
Currently, only Paypal and Bitcoin are accepted as donation platforms.

**Describe the solution you'd like**
Add widely used platforms for giving donations. I recommend open source LiberaPay, as it's commonly used for open source projects.
> The service does not charge a fee on donations aside from that required by its payment processors, instead financing its own activity through donations handled by the service itself in the same way as donations to third-parties are handled.

Some platforms:
- [LiberaPay](https://liberapay.com/) -- [example](https://liberapay.com/k9mail/donate)
- [Ko-fi](https://ko-fi.com/)
- [Open Collective](https://opencollective.com/) -- [example](https://opencollective.com/darkreader#sponsor)
- [Flattr](https://flattr.com/)
- [Github Sponsors](https://github.com/sponsors)
- [BountySource](https://www.bountysource.com/)

Displaying a sponsor button in your repository: [link](https://help.github.com/en/github/administering-a-repository/displaying-a-sponsor-button-in-your-repository) 

Issue is partially copied from [another project](https://github.com/UriahShaulMandel/BaldPhone/issues/170).

c0f, 2020-10-03:
This is timely.  I'm about to close my Patreon account and recreate my donations on Liberapay.

Regards.

aha999, 2020-11-11:
@fr3ts0n I think it would be awesome if you could make AndrOBD project on LiberaPay with screenshots, description etc., currently you have your username set.

aha999, 2021-01-04:
LiberaPay page is not yet in the app.

yschimke, 2021-01-20:
+1 if you add Github sponsors I'll be happy to contribute.

aha999, 2021-05-03:
What about [OpenCollective ](https://opencollective.com/)? it's a big platform and you can donate with direct banking information...