fr3ts0n/AndrOBD #132: FATAL EXCEPTION: main

532910, 2020-10-18:
This happens every time on AndrOBD close (swipe up from the app drawer).

```
E AndroidRuntime: FATAL EXCEPTION: main
E AndroidRuntime: Process: com.fr3ts0n.ecu.gui.androbd, PID: 22708
E AndroidRuntime: java.lang.RuntimeException: Unable to destroy activity {com.fr3ts0n.ecu.gui.androbd/com.fr3ts0n.ecu.gui.androbd.MainActivity}: java.lang.NullPointerEx
ual method 'void java.util.logging.FileHandler.close()' on a null object reference
E AndroidRuntime:        at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:4941)
E AndroidRuntime:        at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:4970)
E AndroidRuntime:        at android.app.servertransaction.DestroyActivityItem.execute(DestroyActivityItem.java:44)
E AndroidRuntime:        at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:176)
E AndroidRuntime:        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
E AndroidRuntime:        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2016)
E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:107)
E AndroidRuntime:        at android.os.Looper.loop(Looper.java:214)
E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:7356)
E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
E AndroidRuntime:        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:940)
E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void java.util.logging.FileHandler.close()' on a null object reference
E AndroidRuntime:        at com.fr3ts0n.ecu.gui.androbd.MainActivity.onDestroy(MainActivity.java:479)
E AndroidRuntime:        at android.app.Activity.performDestroy(Activity.java:8067)
E AndroidRuntime:        at android.app.Instrumentation.callActivityOnDestroy(Instrumentation.java:1341)
E AndroidRuntime:        at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:4926)
E AndroidRuntime:        ... 11 more
```

op3t on lineage 17.1 + microg (on android 10)

fr3ts0n, 2020-12-13:
This error is caused if you don't allow AndrOBD access to the local files on initial startup.

532910, 2020-12-14:
I wouldn't like to give storage access.

closed 2020-12-15 by fr3ts0n

fr3ts0n, 2020-12-15:
Fixed test version V2.2.3 is available here: https://t.me/AndrOBD_dev/229

532910, 2020-12-15:
I'll test it as soon as it becomes available on F-Droid. Avoid KGBgram!

commit 40252933 (2020-12-15): Fix #132: Corrected handling w/o log file permissions