fr3ts0n/AndrOBD #134: No Fault Codes, no Vehicle Info, but there is OBD Date

brianWreaves, 2020-12-13:
**Describe the bug**
 
After successfully pairing with OBD BT device, ECU detected, and connection is confirmed.  Only one screen is displaying data, that's _OBD Data_ screen.
 
On that screen you can see "Number of Fault Codes" is listed as zero.  However, the SRS light is staying on, on the instrument panel. 
 
I know live data is being read as I can see changes in values for some sensors.  As well, the _OBD fault codes_ screen is showing as "Connected"
 
Without the engine running but ignition On, I've tried the "Clear fault codes" feature anyway, but the SRS light never does turn off.  It seems the attempt to clear fault codes is processing, although never does confirm, as the car is running a little more rough the next time I start it.
 
I have tried a few different setting changes within AndrOBD but none of the change has improved the results  e.g. Secure connection, Preferred OBD protocol, ELM adaptive timing, Use last settings, Keep main screen on
 
**To Reproduce**
 
After pairing & connecting, select any of the features and wait for results
 
**Expected behavior**
 
1.  Return data on screens which are blank
2. Ability to view individual fault codes and reset them
 
**AndrOBD Debug log files**
 
Attached file
 
**Important details on your environment:**
 
- Android OS version:  **7.1.1**
- AndrOBD version:  **2.2.2**
- OBD adapter type:  **ELM327 mini**
- Connected vehicle make, type, year, engine size:  **2001 Honda Civic VTi, D17A2**

**Additional context**
 
1.  I have tried portrait & landscape rotations
2.  Both the red (Indicating power) and the green (Indication communication) LEDs turn on in the ELM adapter
3.  Screenshots
 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

4.  Log files
[AndrOBD.log.0.txt](https://github.com/fr3ts0n/AndrOBD/files/5684145/AndrOBD.log.0.txt)

fr3ts0n, 2020-12-13:
Thank you for your detailled report.

It appears that your vehicle adapter responds rather slow to any request.

Many of these Bluetooth Mini adapters are known to work not too well on ISO9141 systems which your vehicle indicates.
* There are several response times >=5 seconds which is indeed very slow.
  * Independent of requested OBD mode
* Your vehicle does not seem to provide any vehicle info data (mode 09)
* Also Reading trouble codes get NODATA responses from the adapter
  * Which is rather unusual at least for OBD mode 03

Based on these slow responses I could find a misbehaviour of the AndrOBD app how it treats mode changes while waiting on slow requests.
I have seen several cases of adapter responses "STOPPED" which indicate that a slow request was aborted by a new request (mode change). 
This intended mode change appears to get lost in the adapter with the STOPPED event and should get re-issued by AndrOBD in this case.

However, 
The OBD data readings indicate that there is NO Trouble code set in the engine ECU, which is the only ECU that responded to the OBD request.
I also would not expect any Fault to be set in the Engine ECU if it is indicated with SRS fault lamp. This probably requires different diagnostics than OBD/OBD2 on the SRS control unit.

closed 2020-12-13 by fr3ts0n

brianWreaves, 2020-12-13:
Thank you for the quick reply!
 
Sorry, I'm pretty ignorant to the in's & out's of Github's features as I'm not a software dev.
 
Does all of this mean the changes made to the code should help this app work with my car & adapter once released?

fr3ts0n, 2020-12-14:
Yes it does. I will create a test version soon. 
However this will not be able to fix the poor performance of your adapter on your car. This just changes the reaction to it and should prevent loss of requested mode changes.

fr3ts0n, 2020-12-15:
Fixed test version V2.2.3 is available here: https://t.me/AndrOBD_dev/229

brianWreaves, 2020-12-15:
Unfortunately I won't be able to help you test this one.  I don't use Telegram and don't know how to install it unless through F-Droid.  I'm just not a dev.

commit 6a71b75d (2020-12-13): fix #134: re-issue request on elm response 'STOPPED'