fr3ts0n/AndrOBD #138: Add beta versions to Github/ Fdroid

aha999, 2021-01-30:
**Is your feature request related to a problem? Please describe.**
Currently, to test a beta version we have to find, download an apk, install it, and if later updating to fdroid version, we have to uninstall this app first.

**Describe the solution you'd like**
Upload alpha/beta releases, apks to Github/Fdroid. Other app makers are doing so, and it's really nice because you can get all app versions from one place where you get them anyways.

- [Example ](https://github.com/k9mail/k-9/releases)of K-9 mail app on Github (pre-release)

- [Example](https://f-droid.org/en/packages/com.fsck.k9/) of K-9 mail app on Fdroid, where the recommended version is the stable one which by deafult installs when installing an app.

fr3ts0n, 2021-02-15:
# Current state

## official versions
- All official versions are released on F-Droid
  - F-Droid offers all mechanisms on update notification, singned app updates, etc.
  - All released versions are signed by F-Droid automatically during the build process (at F-Droid)
  - F-Droid clients tend to have more technical knowledge than i.e. standard PlayStore clients
    - this limits 1st level support requests considerably
  - I am not planning to open another channel of signed versions on any platform which is to be maintained manually ...
- BugFixes shall be released only, if fix was confirmed

## test versions
- All test versions are currently all broadcasted / distributed through the Telegram channel
  - Unsigned test apk's

# Your proposed changes

I have tried to trigger a F-Droid release by creating a GitHub Pre-Release (V2.2.3).

It did trigger the build and update on F-Froid, but also updated the recommended repository version to the latest release. I currently don't know if- and how the k-9 guys have fully automated their releases/pre-releases.
However, even if this can be fully automated, it does not really solve the situation on test versions. 

- Version steps from any test version (back to released/recommended) will require un-/re-installing, even if all versions would be released/signed/shipped from F-Droid platform.

aha999, 2021-05-14:
Maybe @cketti or @eighthave could point us in the right direction ?

cketti, 2021-05-15:
It looks like having F-Droid automatically pick up the latest version will always update the recommended version (`CurrentVersion`), too. See https://f-droid.org/docs/Build_Metadata_Reference/#UpdateCheckMode

K-9 Mail's (beta) releases are manually added to F-Droid by creating a merge request to its [fdroiddata](https://gitlab.com/fdroid/fdroiddata) repository. Example: https://gitlab.com/fdroid/fdroiddata/-/merge_requests/8905

We're doing this manually. But I guess there's no reason why this can't be automated. 

However, a nicer solution would probably be to add something like an `UnstableUpdateCheckMode` to F-Droid. That could add a release but not update `CurrentVersion`, making it an "unstable" version. It would enable projects to e.g. use a different tag scheme for test versions.

fr3ts0n, 2021-05-15:
If I have to release a test version manually, I might as well just provide a test apk on the telegram channel as I do it now.
This is considerably less effort and does not have any delay time to the availability of the apk.

The above mentioned potential solution would need to be integrated into F-Droid framework.

closed 2021-09-12 by fr3ts0n