fr3ts0n/AndrOBD #139: androbd doesn't display data via usb and crashes after changing settings

20210213, 2021-02-13:
With androbd V2.2.3 (20203) under android 9 the following happens:
1. "clear application data" is required to start with.
2. Plug in the usb, confirm to launch androbd, choose usb interface works OK.
3. Connection seems to be established but no data are displayed.
4. After changing any settings (e. g. suppress ATZ, change baud rate), closing the programme and restarting with step 2, android immediately terminates androbd with a corresponding notification. To start again, step 1 is required.
Expected behaviour is that androbd connects via the ELM327 USB OBD II adapter to the CAN bus and displays the data e. g. of a 2009 W169. The last version working this way was com.fr3ts0n.ecu.gui.androbd_20103.apk.
[AndrOBD.log.2.txt](https://github.com/fr3ts0n/AndrOBD/files/5975644/AndrOBD.log.2.txt)
[AndrOBD.log.3.txt](https://github.com/fr3ts0n/AndrOBD/files/5975645/AndrOBD.log.3.txt)

fr3ts0n, 2021-02-13:
Thank you for your report,

Have you tried to start AndrOBD via launcher button first after 'clear application data'?
- This step surprises me, since it is not really necessary from previous V2.2.x versions
- Please also check the app permission storage to be set.

Since there is no log output after the change to CONNECTING state, could you also check with logcat output if there is any more information what happens after that point?

Thx

fr3ts0n, 2021-02-15:
I have found a bug which appears to be in there since V2.1.3 (like you have reported)

Here is a new test version which should fix the problem ...

https://t.me/AndrOBD_dev/230

20210213, 2021-02-17:
Thank you! Unfortunately the packet installer (with rights granted to install from this source) asks whether to install the apk but ends up with an error message that the installation has not been performed.

fr3ts0n, 2021-02-18:
I have verified the apk to be valid. What is the error message you get?
Since this is a unsigned test apk your installer may refuse to install this over a F-Droid signed release version. 
In that case it may require to uninstall the release version first.

20210213, 2021-02-19:
Thank you:
Installation of 2.2.4 indeed worked after removing the previous signed version.
2.2.4 did not crash anymore - the respective bug is obviously successfully fixed.
However it still doesn't display any data (from the car with ignition turned on, and although varying settings like baud rate; log attached). Any idea what might still be wrong?
[AndrOBD.log.0.txt](https://github.com/fr3ts0n/AndrOBD/files/6009595/AndrOBD.log.0.txt)

fr3ts0n, 2021-02-19:
In your log file I can see some commands to be sent to your adapter, but there is never any response coming back from it ...
- USB port seems to be opened successfully
- TX message seems to be sent successfully
- NO response detected

What adapter type is this?

Have you ever tried to talk to your adapter with a terminal program?
Can you see any response there?

20210213, 2021-02-19:
It is some probably Chinese no-name adapter "OBD II Interface USB V1.5".
As a cross-check I uninstalled the debug version androbd 2.2.4 and reinstalled the com.fr3ts0n.ecu.gui.androbd_20103.apk formerly from f-droid. After deleting storage, giving permission to access  storage and disabling ATZ, the old version works, cf. the (shortened) log file attached. 
So the adapter is obviously basically working, but there must be some change in the code after 2.1.3 which prevents the newer apks to run properly on this device. 
[log.txt](https://github.com/fr3ts0n/AndrOBD/files/6011877/log.txt)

fr3ts0n, 2021-02-20:
The update of USB lib did require changes in lib calls.

Here is an updated test version.
https://t.me/AndrOBD_dev/231

Thanks for your re-tests

20210213, 2021-02-21:
That was it - the new test version works out of the box. Thank you!

fr3ts0n, 2021-02-21:
Yes :smiley:  

I'll trigger a release to have the fix available on F-Droid soon...
Thank you for your efforts :+1:

closed 2021-02-21 by fr3ts0n