fr3ts0n/AndrOBD #141: Open web links inside the app

aha999, 2021-04-14:
**Is your feature request related to a problem? Please describe.**
Currently, when for example you hold the diagnostic trouble code, AndrOBD opens a search link by externally opening a default web browser app to display the search for that error code in a new tab, which is personally very annoying, as doing so means we have to close the tab manually and then close the browser and then go back to AndrOBD via app manager switch.

**Describe the solution you'd like**
Open internet links inside the app by accessing the default browser that opens as a pop-up embedded browser. That way, we will be able to see what the error code was, and press X to close that search and go straight back to andrOBD, or what you were viewing. This should be a default behaviour as it would be much more user friendly and convenient, but there should also be a toggle in the settings "Open web links in-app" that would be switched ON by default.

**Additional context**
Apps like Reddit, Discord etc. do this.
Toggle:
[image]

Opened (cropped) website inside the app with pop-up (menu is at the bottom probably because i have it set like this in Firefox), it also has options at the right side like Copy, share, open in main browser...
[image]

Clicking options:
[image]

fr3ts0n, 2021-04-15:
Do you mean something like this?
That is what AndrOBD already does ...

[image]

aha999, 2021-04-16:
No. It should open a default browser's tab.
Just found out how it is called: **Browser Custom tabs** (or Chrome custom tabs).

- [Chrome Custom Tabs](https://youtu.be/QOxIdbNwpx0), [Overview](https://developer.chrome.com/docs/android/custom-tabs/overview/)
- [Best Practices for Custom Tabs](https://medium.com/google-developers/best-practices-for-custom-tabs-5700e55143ee)
- i think this may be a dev site for java browser custom tabs: [androidx.browser.customtabs](https://developer.android.com/reference/androidx/browser/customtabs/package-summary)

aha999, 2021-05-27:
No. You have WebView and it has limited abilities, and isn't as convenient. It should open a default browser's custom tab.

It is called **Browser Custom tabs** (or Chrome custom tabs).

- [Chrome Custom Tabs](https://youtu.be/QOxIdbNwpx0), [Overview](https://developer.chrome.com/docs/android/custom-tabs/overview/)
- [Best Practices for Custom Tabs](https://medium.com/google-developers/best-practices-for-custom-tabs-5700e55143ee)
- i think this may be a dev site for java browser custom tabs: [androidx.browser.customtabs](https://developer.android.com/reference/androidx/browser/customtabs/package-summary)

fr3ts0n, 2021-05-28:
This only uses webview if NO other browser is installed, which is the case on the emulator instance above.
If installed it uses your default browser.

fr3ts0n, 2021-05-28:
I don't really like the idea of implementing something like this, because it breaks at least one of below golden rules
- Functional
- Lean
- Backwards compatible

It would break compatibility with older versions of Android.
I prefer to keep AndrOBD backwards compatible to allow users with older devices also to benefit from functional updates.

aha999, 2021-05-28:
In that case, there should at least be a toggle option to enable Browser Custom Tabs (open web links in-app). 

That way, we wouldn't have to close the tabs and browser / switch applications, because Browser custom tabs have a close button that returns you to the application, i think that is way more convenient and elegant way after viewing the diagnostic code online, and it uses a default browser, all major browsers support browser custom tabs.

fr3ts0n, 2021-05-29:
I don't get the sense of this ...
- The solution would be only available for latest version 
  - NOT COMPATIBLE 
    - -> keep 2 solutions optional 
      - NOT LEAN
      - Support more complex due to functional variants

It may look look more sexy to you to have in-app browse window, but I can not see any functional advantage to the current state.
- Single solution 
  - default browser is opened
  - close browser returns to app (last state)
  - works on ALL versions of Android
    - identical behaviour between different versions

closed 2021-09-12 by fr3ts0n