fr3ts0n/AndrOBD #142: BMW, ELM327 from Lescars, only working intermittently

dpocock, 2021-04-29:
**Describe the bug**
Purchased a new Lescars ELM327 bluetooth.  Installed it in a BMW.  It pairs with the app but it only intermittently works.  It only works for a few seconds, it downloads the vehicle information and then other functions don't work.

I saw a note in the Lescars ELM327 manual "Certains marques, telles que BMW ou Mercedes, proposent leurs propres applications pour la lecture des données".

There are discussions in the BimmerForums forum [where they suggest that ELM327 will never work and it is necessary to use VAG-COM cables](https://www.bimmerforums.com/forum/showthread.php?1928823-Diagnostic-software-for-ELM327&p=25586129#post25586129).  It is not clear if there is any free software solution for VAG-COM.

**To Reproduce**
Steps to reproduce the behavior: - see above

**Expected behavior**
Should work every time.

**AndrOBD Debug log files**
Before I do this, can you please clarify if I am the first person with this problem.

**Log detail level**
The detail level of the logged debug information can be adjusted in the Settings > Development options > Logging level.

By default the detail level is set to INFO.
Preferred detail level for issue analysis is DEBUG. (Since V1.5.0 this level is called FINE)

**Important details on your environment:**
- Lineageos
- v2.2.2 from F-Droid
- Lescars ELM327
- BMW

**Additional context**
Add any other context about the problem here.

fr3ts0n, 2021-05-14:
Thanks for your report.

Could you please provide some log files from the connection attempts.
(for details on logging see: https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues)

Have you tried the latest V2.2.4 already?
There have been some improvements in communication to some (slower) ELM adapters. 
Maybe your problem is already adressed? ;) 

Thx

dpocock, 2021-12-27:
I've sent you a copy of the log by private email

I've tried two different permutations of the device from Lescars:

- ZX-3017-675 (this is the one I was using when I opened the bug report and recorded the log file)
- NX-3014-675 (I got this last week)

They both behave in a similar way, sometimes they lose the connection and try to reconnect many times.

fr3ts0n, 2021-12-30:
Here are my notes on analysis of your log file:

## Notes:
I see a lot of inconsistencies in your log file and I cannot find any obvious cause to that.
- Adapter responses on ATZ vary
  - seen this on cheap chinese clones
  - Possibly disable ATZ in DEV options
    - will be replaced with ATI in sequence
- Inconsistent response to Mode $01/$09 requests
  - "UNABLETOCONNECT", rx with CAN IDs, rx incomplete, rx clean
  - Inconsistent initialisation of vehilce (CAN) bus?
  - may be related to ATZ issue
    
My assumption is that your adapters have a timing issue between Bluetooth (Host) and CAN (Vehicle) protocol 
I have seen similar behaviour on some cheap ELM clone devices before, especially on vehicle using CAN bus.

## Suggestions:
Following workarounds helped most of the useress:
- Disable SECURE connection in OBD options (you already do this)
- Disable ATZ command in Development options

dpocock, 2021-12-30:
I disabled the ATZ command.

I continue to observe similar problems

With ZX-3017-675 the connection is established and I can see the vehicle info and data screens.  After changing between screens a few times, I notice the screens appear blank.  Only the header at the top is visible.  It is always "Connected" but the data is missing

With NX-3014-675 it is not connecting at all, it gets stuck at "ECU detect..." and I don't see anything else.

Is it possible for the app to detect the bad ELM devices and display a warning?

I can buy a different device if necessary, can you make any suggestions on the good devices to buy?  I am in Switzerland, I found these on digitec.ch but I can buy from another supplier or import from Germany or elsewhere if necessary.

fr3ts0n, 2021-12-31:
Did you ever try to use AndrOBD with your adapters on another mobile?
That might be even a matter of combination os/phone/adapter/vehicle

I cannot really suggest a special adapter, but I will initiate an issue / community post for users to post their experience with certain adapters / vehicles /  mobiles.
I hope this might help you and also others to find a known good solution.

fr3ts0n, 2021-12-31:
Please see #187