fr3ts0n/AndrOBD #143: Support for diesel particle filter regeneration

dpocock, 2021-04-29:
**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
support for DPF regeneration, this looks like a useful feature for people who do their own service:
http://www.diagwiki.com/bmw-diesel-particulate-filter-dpf-regeneration

**Describe alternatives you've considered**
other apps

**Additional context**

fr3ts0n, 2021-05-03:
There is support for DPF regeneration specified in the OBD standard. (Mode 08 / TID 02)
However I don't know how many vehicle manufacturers / types do actually support this OBD DPF regeneration.

Is this what you would expect to be implemented from AndrOBD?

dpocock, 2021-05-05:
Yes, the diesel cars are very common in some countries and this is a very important feature for these cars.

https://en.wikipedia.org/wiki/Diesel_particulate_filter#Regeneration

fr3ts0n, 2021-05-14:
Here is a test version which supports OBD Mode $08 test controls. 
- This includes the OBD DPF regeneration.

https://t.me/AndrOBD_dev/232

Unfortunately I could not get my hands on any car which supports any Mode $08 test here, so I have to rely on your testing support now.

Please ensure to generate and report log files of these tests 
(see: https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues#report-issues)
- Please perform "OBD test control" from OBD menu on your vehicle(s).
- You will see test items only if your car supports any OBD Mode $08 test.
- Long-Press any supported/listed test item to confirm- and activate test.

Thx

fr3ts0n, 2021-06-24:
Implemented in >= V2.3.1

closed 2021-06-24 by fr3ts0n