fr3ts0n/AndrOBD #144: [Bug] No Activity found in Bitcoin option

imaryandokani, 2021-04-30:
**Describe the bug**
When I go to settings page and click support project (either paypal or bitcoin), it shows "No activity found to handle intent"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'settings'
2. Click on 'support project' (paypal or bitcoin)
3. See error message "No activity found to handle intent"

[image]

**Expected behavior**
A paypal or bitcoin donating page should be displayed.

**Important details on your environment:**
- Android OS version: Nexus 5X API 24
- AndrOBD version: 2.0.7

fr3ts0n, 2021-05-03:
The link / intent ist supposed to start your bitcoin wallet app.
You will get this message only if you have NO bitcoin wallet installed on your device.

aha999, 2021-05-05:
Which means, that someone who would donate, can't donate it if they can't see the address.
The Bitcoin option should at least show the bitcoin address so users can see/copy what address to donate to... or else it's the only option if you have the wallet app installed.
Or have a drop-down menu on Bitcoin option that displays the number or other options, like See number, copy number...
Or Instead of nothing, show a text "No Bitcoin Wallet installed", so here's the Address: ... ... ... ...
Currently in my opinion this is not a good app behaviour.

fr3ts0n, 2021-05-17:
The thing is that this setting just plainly triggers a intent to open a URL (in this case a bitcoin://... URL)
This kind of intent does not have any specific error handling to be specified.

What I did now is:
- I added the plain text URL as descriptive text to these settings items, so it is at least readable from the settings.

fr3ts0n, 2021-05-17:
Test version: V2.3.2
https://t.me/AndrOBD_dev/234

closed 2021-09-12 by fr3ts0n