fr3ts0n/AndrOBD #149: Self updater with github apk

aha999, 2021-05-12:
**Is your feature request related to a problem? Please describe.**
If we install AndrOBD with directly installing apk from github, we'll have to update it manually.

**Describe the solution you'd like**
If possible, please add automatic update checking/updating to Github apk's.

**Describe alternatives you've considered**
Another open source app, [Baldphone](https://github.com/UriahShaulMandel/BaldPhone), has automatic updater if apk was directly downloaded from github, and if installed from f-droid, self updater was not present.

fr3ts0n, 2021-05-14:
I don't plan to make apk's downloadable / updateable from github.
- This would create potential signaure mismatch with any F-Droid release.
- Especially in combination with plugins this mix could create a complete mess
  - This may cause considerable support effort (which is avoidable)

My rule for this: Release = F-Droid!

aha999, 2021-05-14:
Makes sense

closed 2021-05-14 by aha999