fr3ts0n/AndrOBD #152: No sensor data

Garridus, 2021-05-14:
Hi
I can connect, I get the vehicle information but I don't get data.

[https://drive.google.com/file/d/1I_qUNLTlhlh3e5VQcNEZVq3mre3XyW_d/view?usp=sharing](url)

- Android OS version 11
- AndrOBD version 2.2.4
- OBD adapter type Generic bluetooth
- Connected vehicle make, type, year, engine size- Ford Ka 2008 1.6

Thank you

fr3ts0n, 2021-05-15:
Thanks for your report,

Your OBD adapter produces some interesting responses ... :wink: 
2021-05-14      15:45:51.856    FINE    com.fr3ts0n.prot        ELM rx:'sorry,thistypeisnotsupportedatnow!' (ATSP1)
2
I have never seen a adapter responding **"sorry, this type is not supported at now!"** before.
What adapter is this?
You just mentioned it as a generic bluetooth ...

There is also some strange behaviour in OBD data mode, since the ECU responds with a indication to provide more OBD data items than the below, but when AndrOBD requests for it (0120) the adapter returns NODATA.

However, based on the log file you should have seen at least following PID's in OBD data screen:
[1, 3, 4, 5, 6, 7, 11, 12, 13, 14, 15, 17, 19, 20, 21, 28]

Garridus, 2021-05-15:
Hi

adapter :
https://http2.mlstatic.com/D_NQ_NP_939854-MLA31019644429_062019-O.webp

No data in the screen 

Export:
[https://drive.google.com/file/d/14fSK_txUjDyPr3tflx1EnOmzQYmkyO9C/view?usp=sharing
](url)

I found this data in the mqqtt database.
I don't know if it's from the last connection

[https://drive.google.com/file/d/1cGnYDh74UafGuDgefK0uhe5vqVfIlJ4h/view?usp=sharing](url)

Thank you

fr3ts0n, 2021-05-15:
Hi,
As I stated above, AndrOBD has read multiple PID's in OBD data mode, so the OBD communication part was functional.
If you could not see these data items, is ist possible that you may have set a display filter in settings, which does prevent the app from displaying these?

Default for this filter setting is to display all items, but as soon as it is set manually once, only the marked data items will be shown on the screen.
You can check / modify these filter settings manually in **Settings -> OBD options -> DATA items to display**, but you may have to enable many PIDs in a long list.

As Alternative you can delete the app data in android app settings. This should reset all settings to default, which also deletes any potentially activated filters.

Garridus, 2021-05-15:
All PIDs enabled,   mqqtt only send phone sensor data. I deleted all the data and the problem persists.

fr3ts0n, 2021-05-15:
Ok, just to ge this right.
- Do you use: 
  - AndrOBD V2.2.4? 
  - AndrOBD MqttPublisher V1.0.4?

Which OBD Data is displayed in AndrOBD data screen?

Which OBD data is selected to be published with MqttPublisher?
- There is a similar setting in MqttPublisher to select data to be published

Which OBD data is received by the MQTT broker?

Garridus, 2021-05-15:
* AndrOBD V2.2.4?
YES

   * AndrOBD MqttPublisher V1.0.4?
YES 

> Which OBD Data is displayed in AndrOBD data screen?
nothing

> Which OBD data is selected to be published with MqttPublisher?
ALL 

> * There is a similar setting in MqttPublisher to select data to be published
YES 

> Which OBD data is received by the MQTT broker?

in last  session log nothing

https://drive.google.com/file/d/1I_qUNLTlhlh3e5VQcNEZVq3mre3XyW_d/view?usp=sharing

fr3ts0n, 2021-05-16:
Based on the log file you should be able to see at least 25 data items.
These data items are requested, responded, analyzed, formatted, but as you state not displayed.

Do you have any preferences activated to auto-select previous selections on startup? (Common Settings -> Use last settings)
- Maybe there are some invalid PIDs in pre-selection ...
- This situation can be solved by selecting "Reset last pre-selections" from menu

Other than that, I am running out of ideas right now :cry: 

Well, one more chance :wink: 
- Could you record- and send me a logcat output of your OBD session
  - There might be more helpful information contained than in the logfiles.
    - plugin messaging
    - system display messages/warnings?

fr3ts0n, 2021-05-17:
Test version V2.3.2 is available here:
https://t.me/AndrOBD_dev/234

It contains a potential fix for the behaviour with pre-selections.

ItsiAdam, 2021-05-17:
Did that build fix the issue for you?

I've got the same device and a similar issue, except I've only managed to get the vehicle info to show once. I don't have the "Use last settings" option enabled.

Log: https://nekobin.com/cumunikice

fr3ts0n, 2021-05-17:
@ItsiAdam 
Agreed, please could you create another issue?

Based on the linked logfile the situation looks quite different in your case. ...

Garridus, 2021-05-17:
> Did that build fix the issue for you? (I'll open my own issue if so)
> 
> I've got the same device and a similar issue, except I've only managed to get the vehicle info to show once. I don't have the "Use last settings" option enabled.
> 
> Log: https://nekobin.com/cumunikice

same problem with the new version

fr3ts0n, 2021-05-17:
@Garridus 
Did you check the pre-selection part?

Could you please record- and send me a logcat output of your OBD session
- There might be more helpful information contained than in the logfiles.
  - plugin messaging
  - system display messages/warnings?

darekdeo, 2021-09-01:
Same issue here on:
SM-G970F
Android 11
Audi A4 B6 2001

I think I will have time this weekend to compile the app and debug it in my car.

darekdeo, 2021-09-04:
> Same issue here on:
> SM-G970F
> Android 11
> Audi A4 B6 2001
> 
> I think I will have time this weekend to compile the app and debug it in my car.

I've compiled debug version from master (135a0456f5952a763c55379f612c45a3375b6d8a)

This time AndrOBD did display OBD Data. Both from DEMO and from REAL car.

At home I've also compiled release variant and checked if at least DEMO data is displayed, it is.

The only thing I couldn't get was OBD test control (in REAL car), I've received "service 0x08 not supported androbd" (ElmProt.handleTelegram received in buffer: 7F0811). Other app (CarScanner) is able to read DTC error codes, is it the same functionality as OBD test control?. If so, I believe this subject could be reported in another issue.

Oh, and Android device and car were the same as reported before. I've forgot previously to report model of adapter, here it is: Vgate iCar2 (https://vgate.pl/product/vgate-icar2/)

fr3ts0n, 2021-09-06:
HI @darekdeo ,
Thanks for your report.

OBD service 08 (Test control) is a different service than reading DTCs, but it is not supported by many vehicles. If supported by verhicle it would offer Actions like "DPF regeneration" or "Evap system leak test".
OBD DTC codes can be read by Androbd. (OBD sevices 03-DTCs, 07-pending DTCs, 0A-permanent DTCs)

closed 2021-09-12 by fr3ts0n