fr3ts0n/AndrOBD #156: Some strings are not in Weblate. Please update.

aha999, 2021-05-14:
**Describe the bug**
Strings like "Identify plugins", "OBD test control","Evaporative tsystem leak test", "Diesel Particulate filter regeneration" and others are not in Weblate to translate.

**Describe solution**
Update the strings so all of them will be translatable in Weblate.

**Important details on your environment:**
- Android OS version 11
- AndrOBD version 2.3.1 (Test version)
- OBD adapter type Demo

fr3ts0n, 2021-05-15:
OBD test control and the new OBD test description are translatable on weblate.
These strings are located in the platform independent OBD library and therefore translated in the OBD data section

There are several strings from **plugin** module which are currently not mirrored to weblate for translation

fr3ts0n, 2021-05-15:
Now also AndrOBD plugin framework strings are translatable in weblate

closed 2021-05-15 by fr3ts0n