fr3ts0n/AndrOBD #165: Sensors - Suspension

stef204, 2021-07-16:
I am trying to get values for the rear suspension leveling sensors on my vehicle which has an "air" suspension.
I imagine this is out of the scope of AndrOBD?
Probably due to proprietary software from manufacturer?
Curious to understand more.

fr3ts0n, 2021-07-17:
The OBD protocol is mandated by government to be supported by every road vehicle, but only for control units which are relevant for exhaust emissions.
The OBD protocol and data mapping is standardized for all vehicles, unlike all other diagnostic protocols and data items (like suspension data also) which are specific to the vehicle OEMs / types, 

AndrOBD does support the OBD protocol only, and therefore is limited to emissions related data as well.

- Any support for additional diagnostic data within AndrOBD would:
  - Require OEM internal documentation about diagnostic data mapping and scaling
    - and/or a huge amount of time for reverse engineering, and testing of these
  - Require to change to a full time, closed source, pay app project, to pay off above efforts.
  - Make the app OEM specific (at least extensions)

Since I prefer AndrOBD development to stay a open source spare time project, I don't see a chance to implement any additional protocols / ECU data.

stef204, 2021-07-18:
Understood, thanks for the detailed feedback.

closed 2021-07-18 by stef204