fr3ts0n/AndrOBD #166: mostly wiki

theduskin, 2021-07-19:
**Is your feature request related to a problem? Please describe.**
i do not fully understand the basic concept. what to do/where to start with a new or known vehicle.
i understand, if there is 3 times no response for a pid, it is excluded from request.
how is that related to settings-obd-daten zur anzeige?
is there a config file for "daten zur anzeige"?
when i choose some pids for Graph display, are these the only requested pids or grabs it always the whole bunch?

**Describe the solution you'd like**
i would like to set up a vehicle or problem specific configuration on laptop/pc and save it for reuse.
and also a report about the supported pids.

fr3ts0n, 2021-07-19:
Hi @theduskin 

In general AndrOBD does only request the PIDs which are selected for display.

After initial start of a OBD communication session, this list contains all PID's  which the vehicle reports to be supported.

The configuration settings "obd-daten-zur-anzeige" can be used to remove certain OBD data items from data display globally for every vehicle. This setting defaults to *show all*
Any data item which is supressed from display by this setting will *NOT* be requested.

The next step of PID filtering can be applied at runtime by selecting >0 data items in OBD data screen and then select display function
- Gauge
- HUD
- Graph
- Filtered List

Also this selection limits the requested PID's to the selected data items only.

For purpose of automation above runtime selections of
- ELM adapter 
- ECU connection
- OBD service/function
- Data items
- Display mode

can be automatically preselected for the next program start.
This startup behaviour can be configured in Settings -> Common -> Use last settings
These remembered last settings can be easily forgotten by "Reset last preselections" from menu. This is usually done if you prepare to connect to a new vehicle.

Finally to the mechanism you have mentioned above:
Some vehicles do erraticaly indicate some PIDs as supported, but when requested these PID requests are responded with *NO DATA*. 
Since this behaviour could be caused temporarily by the [Adaptive timing feature](https://github.com/fr3ts0n/AndrOBD/wiki/Data-item-update-time#adaptive-timing-feature), the erratic PID will be disabled after 3 consecutive errors.

fr3ts0n, 2021-07-22:
For saving / restoring app configurations to / from a pc I would suggest to use the ADB tool
( see https://stackoverflow.com/questions/1243079/how-to-examine-sharedpreferences-from-adb-shell )

theduskin, 2021-07-23:
Thank you Erwin,
your explanation helps me.

i think about a little tool to get the prefs, get the real names of the data_items set from pids.csv,
and a selection list to generate a prefs.xml.
would you mean this is save for the future or not worth the effort (if you plan to change the structure)?

Danke und Gruß
Christoph

fr3ts0n, 2021-07-23:
Currently there is no intention to change the data model.
However, there is no guarantee if there comes a change which requires a model change ...

theduskin, 2021-07-27:
ok, ive done it and its working so far.
root for adb is needed, and i have to kill the app before pushing the xml.
[image]

closed 2021-09-12 by fr3ts0n