fr3ts0n/AndrOBD #167: MQTT and GPS plugins not available on F-Droid

camsellem, 2021-07-26:
Hi,

I can't find the plugins for the GPS Sensor and the MQTT Publisher. Were they discontinued?  Thanks

[image]

fr3ts0n, 2021-07-28:
Yes, I know, there is a problem with the metadata links (icons and descriptions) on fdroid web site.

There are all available extension packages listed with identical description / icon.
If you mouseover each package ypu will find each entry to link to a different apk:
- gps provider
- mqtt publisher
- sensor provider

So you can download the desired apk, but currently you will have to trace through the different links.

aha999, 2021-08-21:
This is really confusing also for me... If this is caused by metadata links, I think AndrOBD should change to Organization on Github and have separate repos for every plugin. It would also sort other things out nicely... Like if for example in the future AndrOBD gets ported to iOS or other devices, they could be easily findable under AndrOBD organization :)

fr3ts0n, 2021-09-12:
Fixed on F-Droid:
[image]

closed 2021-09-12 by fr3ts0n