fr3ts0n/AndrOBD #173: False values O2-Lambda-Sensor

Artur-debug, 2021-09-17:
[image]

Hi, how is the lambda value O2 sensor calculated? I think the app miscalculates this value (Service 01 PID 24). The values that AndrODB shows me do not match a turbo diesel and other.

It looks like AndrOBD calculates the lambda value (Service 01 PID 24) like this:

EQ_RAT = 2/65535 * Byte_A_B_PID24

However, if Byte_A_PID4F (Service 01 PID 4F) is set, then this must be included in the calculation:

EQ_RAT = Byte_A_PID4F / 65535 * Byte_A_B_PID24

For me the Byte_A_PID4F = 32_ (10) and therefore has to multiply the AndrOBD values by 16 so that it fits.

[image]
[image]

**Important details on your environment:**
- Android OS version 10
- AndrOBD version 2.2.4
- OBD adapter type 
- Audi A3 Sportsback (8PA) 2005 2L TDI 140 PS

Artur-debug, 2021-09-17:
Additional info:

>ATZ
ELM327 v1.5
>0120
41 20 90 00 00 01

Artur-debug, 2021-09-17:
Small supplement:
The picture above was taken with the engine idling. 16 * 0.217 would result in a lambda value of 3.472. I would say that this is a normal value for turbo diesel. The values can also be far higher in idle mode.  
Engine at full load, AndrOBD measures values around 0.08 and would produce a correct lambda value of 16 * 0.08 = 1.28. In the range Lambda 1.05 - 1.5, turbo diesels work at full engine load according literature.

fr3ts0n, 2021-09-18:
HI @Artur-debug 
Thank you for your detailled report.

You are absolutely correct. 
Currently AndrOBD does only use a static calculation for PID's $24... data. 
The values of PID $4F are not considered yet.

I'll try to implement this and inform you when a test version is availabe.

fr3ts0n, 2021-09-19:
Test version (V2.4.2) is available here for your tests:
https://t.me/AndrOBD_dev/235

fr3ts0n, 2021-09-19:
Sorry, I didn't mean to close the issue before you can confirm the BugFix ...

Artur-debug, 2021-09-19:
Works great.

In the picture you can see the lambda warm-up phase. With broadband probes, the lambda value is generated by a pump current in the probe. 0 mA pump current means lambda = 1. During the warm-up phase the pump current is also 0 mA and shows me lambda = 1. The later values also fit. 

[image]

Artur-debug, 2021-09-26:
Often the first two values do not go into the new calculation with the new scaling factor 32. 

0.062 x 16 = 0.992

[image]

fr3ts0n, 2021-09-27:
Looks like these are requests of PID $24 before PID $4F got requested and conversion factors got updated
Currently there is no way to define dependencies between PIDs and ensure requests in order of dependencies.

commit 382fdab4 (2021-09-19): Fix #173 - False values O2-Lambda-Sensor

- Add mnemonic table to EcuDataItems
- Add optional dynamic factor to LinearConversion
- Reference PID data for dynamic factor in conversion
closed 2021-09-27 by Artur-debug