fr3ts0n/AndrOBD #174: Byte misinterpretation of mode 09 PID 00

Artur-debug, 2021-09-21:
In the picture the blue byte should be a Message_Counter_Byte. AndrOBD interprets this byte as data and thinks that the PID 08, 09, 0a, 0b, 0c are supported.
My car only supports PID 01, 02, 03, 04. That should be the first data byte F0 after the Message_Count_Byte. 

[image]

**Important details on your environment:**
- Android OS version 10
- AndrOBD version 2.4.2
- OBD adapter type 
- Audi A3 (8PA) 2005 2L 140PS

[Mode09PID00_ByteMisinterpretation.txt](https://github.com/fr3ts0n/AndrOBD/files/7206454/Mode09PID00_ByteMisinterpretation.txt)

Artur-debug, 2021-09-21:
[image]

Not every PID seems to have a Message_Count_Byte. 

>0900
SEARCHING...
49 00 01 F0 00 00 00

>0901
49 01 05

>0902
49 02 01 00 00 00 xx*
49 02 02 xx xx xx xx
49 02 03 xx xx xx xx
49 02 04 xx xx xx xx
49 02 05 xx xx xx xx

*xx is Data

>0903
49 03 04

>0904
49 04 01 xx xx xx xx
49 04 02 xx xx xx xx
49 04 03 xx xx xx xx 
49 04 04 xx xx xx xx 

>0905
7F 09 12 

>0906
7F 09 12 

>0907
7F 09 12 

>0908
7F 09 12 

>0909
7F 09 12 

>090a
7F 09 12

fr3ts0n, 2021-09-23:
Thanks again for the next very clear report,

Bug confirmed!
The message count appears to be optional, probably dependent on the vehicle implementation, since I do have many cases indicating the mode 9 pid's without message count ...
However, obviously it needs to be treated optionally

I'll try to find a solution and then post a fix for your testing / verification

fr3ts0n, 2021-09-23:
Test version (V2.4.3) is now available here:
https://t.me/c/1434285001/1138

Artur-debug, 2021-09-24:
Length of VIN should be 17. Number of VIN should be 1. Length of CAL-ID should be 16 and the number of CAL-ID should be 1.

[image] 

[AndrOBD.log.0 (6).txt](https://github.com/fr3ts0n/AndrOBD/files/7225926/AndrOBD.log.0.6.txt)

[image]

Artur-debug, 2021-09-24:
[image]

Artur-debug, 2021-09-24:
[image]

Artur-debug, 2021-09-25:
Probably the ISO standards are different.

Artur-debug, 2021-09-26:
After App-Update 0902 is not queried :

[image] 

2021-09-26	15:42:59.964	FINE	com.fr3ts0n.prot	ELM rx:'>' (0900)
2021-09-26	15:42:59.965	FINE	data.ecu	01 Länge der Fahrgestellnummer (VIN)            240.0 
2021-09-26	15:42:59.966	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@72ca631 TX:'0900'
2021-09-26	15:43:00.105	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@72ca631 RX:'490001F0000000'
2021-09-26	15:43:00.107	FINE	com.fr3ts0n.prot	ELM rx:'490001F0000000' (0900)
2021-09-26	15:43:00.306	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@72ca631 RX:'>'
2021-09-26	15:43:00.308	FINE	com.fr3ts0n.prot	ELM rx:'>' (0900)
2021-09-26	15:43:00.309	FINE	data.ecu	01 Länge der Fahrgestellnummer (VIN)            240.0 
2021-09-26	15:43:00.311	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@72ca631 TX:'0900'
2021-09-26	15:43:00.448	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@72ca631 RX:'490001F0000000'
2021-09-26	15:43:00.450	FINE	com.fr3ts0n.prot	ELM rx:'490001F0000000' (0900)
2021-09-26	15:43:00.650	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@72ca631 RX:'>'
2021-09-26	15:43:00.652	FINE	com.fr3ts0n.prot	ELM rx:'>' (0900)
2021-09-26	15:43:00.654	FINE	data.ecu	01 Länge der Fahrgestellnummer (VIN)            240.0 
2021-09-26	15:43:00.655	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@72ca631 TX:'0900'
2021-09-26	15:43:00.792	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@72ca631 RX:'490001F0000000'
2021-09-26	15:43:00.794	FINE	com.fr3ts0n.prot	ELM rx:'490001F0000000' (0900)
2021-09-26	15:43:00.994	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@72ca631 RX:'>'
2021-09-26	15:43:00.997	FINE	com.fr3ts0n.prot	ELM rx:'>' (0900)
2021-09-26	15:43:00.999	FINE	data.ecu	01 Länge der Fahrgestellnummer (VIN)            240.0 
2021-09-26	15:43:01.001	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@72ca631 TX:'0900'
2021-09-26	15:43:01.136	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@72ca631 RX:'490001F0000000'
2021-09-26	15:43:01.137	FINE	com.fr3ts0n.prot	ELM rx:'490001F0000000' (0900)
2021-09-26	15:43:01.338	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@72ca631 RX:'>'
2021-09-26	15:43:01.340	FINE	com.fr3ts0n.prot	ELM rx:'>' (0900)
2021-09-26	15:43:01.341	FINE	data.ecu	01 Länge der Fahrgestellnummer (VIN)            240.0 

[AndrOBD.log.0.0101.txt](https://github.com/fr3ts0n/AndrOBD/files/7231499/AndrOBD.log.0.0101.txt)

fr3ts0n, 2021-09-26:
OK my last attempt to remove ISO string padding character did remove valuable information from PID bitmask ...
I really hate these protocol inconsistencies (i.e.: who does conditionally 0-pad a string at it's beginning?)
-> Don't let politics get involved into technical specs ;-)

fr3ts0n, 2021-09-26:
Should be fixed with this update:
https://t.me/AndrOBD_dev/240

Artur-debug, 2021-09-26:
Works fine, thank you!

closed 2021-09-26 by Artur-debug