fr3ts0n/AndrOBD #175: Add Battery Cranking Test

Artur-debug, 2021-09-22:
Please add Battery Cranking Test like this:

[image]
:
Perhaps the ATRV command with a high baud rate is sufficient for this.

fr3ts0n, 2021-09-24:
Isn't this what you already can do just by graphing out the corresponding PID for battery voltage?
- Key on
- Select PID for graph display
- Crank engine
- Check graph

Artur-debug, 2021-09-24:
My vehicle and I think most of them do not support battery related PIDs. I think there is only something for hybrid vehicles.

Even if battery PIDs were supported, one would have to react very quickly in diesel vehicles with a glow plug system. Because the glow plugs start working in the first ignition position.

OBD plugs have pins 16 (battery +) and 4/5 (vehicle / signal ground) which always carry the battery / generator voltage. Even when the ignition is off.

The ATRV command can always be used to measure the battery / generator voltage (if the ELM does not display the correct voltage, it can be calibrated with ATCV).

Before connecting to the ECU, the cranking test can be carried out. A high baud rate would probably be necessary to release the voltage at the time of start-up. 

It would look something like this: 

[image]

Artur-debug, 2021-09-24:
f the Y-axis adjusts itself dynamically then the generator can be checked for correct function in the later process. Multimeters are not needed and the car hood does not have to be opened: 

[image]