fr3ts0n/AndrOBD #178: Add P0000 in $03 if no error code is stored

Artur-debug, 2021-09-26:
Please add P0000 if no error code is stored. So I can immediately see that there is communication and that ELM has received an answer from the ECU. 

[image]

fr3ts0n, 2021-09-26:
Mode $3 should already display a "No codes set" entry if there is no code responded by the vehicle.
Could you please attach a log file of this action?
I would like to verify why your device doesn't show this message.
Thx

Artur-debug, 2021-09-26:
[image]

[AndrOBD.log.0.0900fix.txt](https://github.com/fr3ts0n/AndrOBD/files/7231980/AndrOBD.log.0.0900fix.txt)

closed 2021-09-29 by Artur-debug