fr3ts0n/AndrOBD #181: False values in dashboard mode, false scaling

Artur-debug, 2021-09-30:
Lambda should be scaled dynamically too.

[image] [image] 

[image] [image] 

This values should be:

[image] [image] 

[image] 

[AndrOBD.log.0.txt](https://github.com/fr3ts0n/AndrOBD/files/7261986/AndrOBD.log.0.txt)

**Important details on your environment:**
- Android OS version 10
- AndrOBD version 2.4.3
- OBD adapter type
- Audi A3 (8PA) 2005

fr3ts0n, 2021-12-13:
Test version available: https://t.me/AndrOBD_dev

Artur-debug, 2021-12-13:
Works good!

fr3ts0n, 2021-12-15:
Released with [V2.5.2](https://github.com/fr3ts0n/AndrOBD/releases/tag/V2.5.2).
will be available on F-Droid soon ...

closed 2021-12-15 by fr3ts0n