fr3ts0n/AndrOBD #185: Do not hide system bars while app UI is displaying

mvglasow, 2021-11-19:
**Is your feature request related to a problem? Please describe.**
I find it somewhat irritating when the notification bar disappears upon displaying AndOBD. It is not an app that would require a particularly immersive UX (such as, say, a video player), so there is no real need to have this. Other apps with similarly low immersiveness requirements also do not hide the notification bar, and the screen real estate gained by hiding the bar is minimal.

**Describe the solution you'd like**
Keep displaying all system bars when AndrOBD is being displayed.

**Describe alternatives you've considered**
If someone really, really wants that immersive experience, not to be disturbed by a bar showing the time, device status and notifications, that behavior could be made configurable.

**Additional context**
Add any other context or screenshots about the feature request here.

fr3ts0n, 2021-12-13:
configurable
Test version available: https://t.me/AndrOBD_dev

fr3ts0n, 2021-12-15:
Released with [V2.5.2](https://github.com/fr3ts0n/AndrOBD/releases/tag/V2.5.2).
will be available on F-Droid soon ...

closed 2021-12-15 by fr3ts0n