fr3ts0n/AndrOBD #186: Displaying o2 voltage gauge use the wrong scale

willhubs, 2021-11-25:
When I am displaying an o2 sensor voltage as a gauge it shows a range of 0-1275mv even though it can never go above 100
Should be able to default this to 0-1000mv or at least be able to adjust the range for each one.

willhubs, 2021-11-25:
[image]

fr3ts0n, 2021-11-25:
The limits are defined bythe resolution/conversion of the OBD protocol.
For these PIDs data is sent as a 8-Bit value (0..255) where each bit represents 5mV, which results in a measurement range of 0..1275 mV. This is what the display range defaults to.

I understand that most O2-sensors can only deliver voltages from 0..1000 mV. However I am not really sure if there are any setups possible which may deliver voltages >1000 mV. 
Limiting the display range for these PID's globally would invalidate the display for any system delivering >1000 mV.

willhubs, 2021-12-04:
Ok, not sure of any other systems that would use anything over 1v either, but an option to customize the display would be awesome.

Also, is there some kind of buffering that happens to slow down the needle movement, because the TPS doesn't move as quickly as it should.

fr3ts0n, 2021-12-13:
Data display customization introdued with V2.5.0
Test version available: https://t.me/AndrOBD_dev

fr3ts0n, 2021-12-15:
Released with [V2.5.2](https://github.com/fr3ts0n/AndrOBD/releases/tag/V2.5.2).
will be available on F-Droid soon ...

closed 2021-12-15 by fr3ts0n