fr3ts0n/AndrOBD #187: Experience with specific vehicles / adapters / mobiles

fr3ts0n, 2021-12-31:
Hi community,

Based on some user requests / issues I would like to initiate a information exchange about your user experience to use AndrOBD on certain vehicles, adapters mobiles.
Some of us do have problems using certain adapters for certain verhiles.

This query could help many others to find good working adapters for their vehicles.

Please post comments with following information
- Vehicle (make, model, year, engine)
- OBD adapter (make, model, product link)
- Mobile (OS, version)
- Your experience
- Rating of usability ⭐⭐⭐⭐⭐

fr3ts0n, 2022-01-07:
OBD adapter: [Mini ELM327 Scanner](https://www.ebay.at/itm/353699689364?hash=item525a24f394:g:zW4AAOSw1G1ZqqOu)

Vehicles: 
- Ford Focus ~2006, 
- Hyundai i10 ~2013, 
- Hyundai i30 ~2015, 
- Nissan Micra ~2002
- ...

Mobile: Various No-Names (Android 7 - 11)

Experience:
Adapter was sold as KKMoon at time of ordering (2015). 
Since it is so tiny, I always have it in my backpack ...
- Adapter works surprisingly good on vehicles with CAN system
- Rather bad on ISO9141 systems (Nissan Micra 2002)
  - Manual pre-selection of ISO protocol did help, but poor performance / response times

Rating of usability: 
- CAN systems: ⭐⭐⭐⭐
- ISO9141 systems: ⭐

fr3ts0n, 2022-01-07:
OBD adapter: [KOEY OBD2 Diagnosegerät](https://www.amazon.de/KOEY-Diagnoseger%C3%A4t-Bluetooth-Adapter-Codeleser/dp/B09NM5X2QB/ref=sr_1_60?crid=1GISNXL2UCOIU&keywords=obd2+bluetooth&qid=1641545719&sprefix=%2Caps%2C161&sr=8-60)

Vehicles:
- Ford Focus ~2006
- Hyundai i10 ~2013
- Hyundai i30 ~2015
- Nissan Micra ~2002
- ...

Mobile: Various No-Names (Android 7 - 11)

Experience:
- Adapter was sold as "Tinxi Bluetooth EOBD OBD 2" at time of ordering (2012).
  - was around 30€
  - I didn't open it up, but I assume it contains a real ELM327 chip
- Works well on all vehicles / protocols I got my hands on

Rating of usability:⭐⭐⭐⭐⭐

SylwesterZarebski, 2022-02-22:
Vehicle: **Ford Focus MK3 FL (2014-2018)**

OBD adapter: **Viaken VSCAN STN1170** (Scantool FTDI drivers)

Mobile: **Android 11 (Samsung One UI 3.1)**

Your experience:
- App do not read modules errors, it did not show any error. Probably because of two CANs (HSCAN and MSCAN) in newer Fords.
- It read some PIDs, but it looks like only from PCM.

Rating of usability ⭐⭐

reedrs, 2022-04-07:
- Vehicle: 1998 Subaru Legacy GT, EJ25 engine (also briefly tested with a few other vehicles with similar behavior)
- OBD adapter: ([Veepeak Mini WiFi OBD II Scanner](https://www.amazon.com/gp/product/B00WPW6BAE))
- Mobile: Oneplus 5T, LineageOS 18 (Android 11)
- Experience:

Connection with ECU drops almost immediately after it begins. Sometimes it stays connected long enough to get a list of trouble codes, but it takes a few tries, and it always disconnects after a short time. This makes looking at any real-time data like RPM impossible.

Also, the WiFi-based adapter is a pain to connect to, because Android judges its WiFi network to have "no internet". I have to wait for a "no internet" notification to pop up, then click "yes, stay connected", otherwise AndrOBD's attempts to connect to 192.168.0.10 get routed to my phone's LTE connection, even while my phone is connected to the device's WiFi network.

- Rating of usability :star:

donpdonp, 2022-06-30:
OBD Adapter: Mini ELM 327
https://www.aliexpress.com/item/2251832764943250.html?spm=a2g0o.order_list.0.0.2a521802V6qndn

Phone: Pixel 5a

The phone will bluetooth-pair with the adapter. That part seems to go fine. When the app starts, the OBDII device shows up on the list of devices, but when I select that device, the app tries to connect for a few seconds, then gives up. No error messages.

Petemir, 2023-04-19:
* Vehicle: Nissan Micra K13 2011
* OBD adapter: [Veepeak Bluetooth 4.0](https://www.amazon.de/-/en/gp/product/B073XKQQQW/ref=ppx_od_dt_b_asin_title_s01?ie=UTF8&psc=1)
* Mobile: Android 12
* Your experience: I could do what I intended to do: clear the engine light for an error that has been checked by 3 different mechanics and could not be solved (and doesn't impact in any way the driving):
* Rating of usability: :star::star::star::star::star:

slender100, 2023-08-27:
Vehicles:
2019 Ford Ecosport
1998 Ford Explorer

OBD Adapter: [FIXD Blueetooth OBD Scanner](https://www.amazon.com/FIXD-Professional-Bluetooth-Reader-Android/dp/B013RIQMEO/)

Mobile: Pixel 3a, CalyxOS (Android 13 Custom Rom)

Experience: I was able to see pretty much everything about the Ford Ecosport, but no vehicle info for the Ford Explorer, which was most likely due to it being 2 years after OBD became a standard, and is most likely not capable of displaying vehicle info. However, this device did show me OBD information, as well as codes for both vehicles, despite their large age gap.

Rating of usability: ⭐⭐⭐⭐⭐

slender100, 2023-09-02:
Vehicles:
2019 Ford Ecosport
1998 Ford Explorer

OBD Adapter: [BAFX Products OBD2 Scanner](https://www.amazon.com/gp/product/B005NLQAHS)

Mobile: Pixel 3a, CalyxOS (Android 13 Custom Rom)

Experience: This scanner can be slow at first, and can sometimes not show data unless the section you are trying to view is re-selected, but works great. This scanner may also be difficult to plug into some OBD2 ports depending on the vehicle. For example, the scanner had to be jiggled onto the OBD2 port of the Ford Ecosport, but only had to be pushed into place by just a bit on the Ford Explorer. Overall, I would recommend this if you are wanting to get a product that is quality controlled when compared to the other cheap chinese ELM327 scanners.

Rating of usability: ⭐⭐⭐⭐

slender100, 2024-01-29:
Vehicles:
2019 Ford Ecosport
1998 Ford Explorer

OBD Adapter: Generic OBD2 ELM327 "Branded" Adapters

[image]

Mobile: Pixel 3a, CalyxOS (Android 13 Custom Rom)

Experience: This scanner works well. Virtually no delay or issue with connectivity. This scanner is also very lightweight, as the pcb is smaller, yet still works as good as others. This scanner typically can go for really cheap on ebay or aliexpress, around 5 USD or less. Be sure to look for scanners of this chassis type, and general look.

Rating of usability: ⭐⭐⭐⭐⭐

slender100, 2024-02-07:
Vehicles:
2019 Ford Ecosport
1998 Ford Explorer

OBD Adapter: "HHOBD Advanced" Adapter

[image]

Mobile: Pixel 3a, CalyxOS (Android 13 Custom Rom)

Experience: This scanner isn't too good. When I tried connecting to the Ecosport's ECU, it would keep having issues connecting until I fully turned the vehicle on. This scanner would refuse to connect to the Explorer completely. Avoid.

Rating of usability: ⭐⭐