fr3ts0n/AndrOBD #188: library for use / testing on Linux?

dpocock, 2022-01-06:
**Is your feature request related to a problem? Please describe.**
Unit testing, compatibility testing

**Describe the solution you'd like**
Can you provide some of your code as a standalone library that can run on Linux without Android?  I understand that the UI depends on Android but maybe some or all of the code for communicating with the car could run headless in Linux and other platforms that have Java.  This might provide more opportunities for other developers to work with your code in their target platform.

**Describe alternatives you've considered**
N/A

**Additional context**
Using OBD2 as a library with Apache Camel would be interesting.  I contributed the Ganglia module there.
https://camel.apache.org/components/3.12.x/ganglia-component.html

fr3ts0n, 2022-01-07:
That's exactly what the sub-directory "**_library_**" of this project is:
- Target independent
- OS independent
- JRE compilable

Actually, this was even the start point of AndrOBD, since I started Java / OBD development with a JDK/Swing application under Linux / Windows.

You can find these Swing applications in package "com.fr3ts0n.ecu.gui.application"... (i.e.: ObdTestFrame.java, ...)

closed 2022-01-07 by fr3ts0n

dpocock, 2022-01-07:
Would you be able to publish the library in Maven Central?

I did a search there like this:
https://search.maven.org/search?q=com.fr3ts0n.ecu

It finds another library, de.valtech.aecu