fr3ts0n/AndrOBD #196: AndrOBD doesn't connect to a vehicle

aha999, 2022-04-21:
**Describe the bug**
andrOBD doean't connect to a Car with Vlink iCar Pro. It connects to the adapter, but it dosconnects.

**To Reproduce**
Steps to reproduce the behavior:
1. Adapter inserted, car contact on
2. try to connect with androbd
3. It says "ECU Detect..., Disconnected, Connecting, Disconnected", and it repeats.

**Expected behavior**
Program should connect to the adapter and car.

**AndrOBD Debug log files**
Wait so i get to my pc and upload (edit) it

Please attach AndrOBD debug log files.

AndrOBD stores log files in a subdirectory on your primary storage device. (com.fr3ts0n.ecu.gui.androbd/log/AndrOBD.log.*)

**Log detail level**
The detail level of the logged debug information can be adjusted in the Settings > Development options > Logging level.

By default the detail level is set to INFO.
Preferred detail level for issue analysis is DEBUG. (Since V1.5.0 this level is called FINE)

**Important details on your environment:**
- Android OS version 11
- AndrOBD version 2.5.2
- OBD adapter type Vgate iCar Pro, Bluetooth 4.0 BLE
- Adapter product website: [link](https://www.vgatemall.com/productinfo/552033.html)
- Connected vehicle Citroen, Berlingo, 2005, 2.0 Diesel

**Additional context**
I assume it's the car's fault, because it also doesn't connect with both proprietary OBDLink MX and ObdLink app.

fr3ts0n, 2022-04-22:
It really looks like your car doesn't talk OBD protocol.

Within the log file, on the OBD connection attempt to your vehicle the adapter reponds UNABLETOCONNECT.
This means it tried to contact the ECU using all known OBD transport protocols (CAN, ISO141 ...), but didnt't get any valid response from the vehicle.

aha999, 2022-04-22:
ok, maybe i will figure out why.