fr3ts0n/AndrOBD #203: APK Release Build?

beadon, 2022-05-03:
**Is your feature request related to a problem? Please describe.**

It would be fantastic to have an APK built automatically on the release page.  This would help anyone poking around a chance to try a pre-built APK.

**Describe the solution you'd like**
Provide a pre-built APK as a download option.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

beadon, 2022-05-03:
Whoop - I see it's available via FDroid!  Perhaps a mention on the releases page to direct users to FDroid then is a quick solve for my poor reading skill?

aha999, 2022-05-03:
The link for F-Droid and beta versions is already in the main readme page here as a button: https://github.com/fr3ts0n/AndrOBD#install and on the landing page here: http://fr3ts0n.github.io/AndrOBD/

closed 2022-07-31 by fr3ts0n