fr3ts0n/AndrOBD #207: What is VGate's relation to this project, and does the ad revenue support development of AndrOBD?

sudofox, 2022-06-18:
Title

I'm assuming Vgate is run by you guys?

If I support the AndrOBD project, can I get the ads removed in the app?

The version button in the app goes to this repository.

https://play.google.com/store/apps/details?id=com.minisatip.vgate

sudofox, 2022-06-18:
[image]

fr3ts0n, 2022-06-19:
Hi, 
Thank you for the info.
No. Vgate ist NOT operated by us. It seems to be the next issue of Code robbery.
These guys just think open source is free of licence. 
I'll start another Google license infringement issue.

sudofox, 2022-06-19:
Aw man, I've been using it for a while. :( I wonder how many there are

sudofox, 2022-06-19:
Actually come to think of it I think when I bought an obd dongle it came with a piece of paper recommending I install the Vgate app because I don't see how I'd have found it otherwise

sudofox, 2022-06-19:
[image]
It looks like they've got a ton of these on sale and also ripped off AndrOBD

They probably have gotten a _lot_ of money from selling hardware with a copy of this app

ieffuie4, 2024-07-05:
Its great to see, that this terrible clone is gone from the closed source google play store.
Maybe for future: Do not buy any vgate products and tell all other people to not buy those. Those company does not respect OpenSource licenses and thus no one who like OpenSource software should spend any money on this company.

Maybe add a detection in AndrOBD to detect vgate adapters and block the functionality with vgate? But also add a one line 'disable vgate blockage' that can be set from 'false' to 'true'. And when someone reports here a not working vgate with a error message that vgate is not supported because of issue #207 , you tell them to compile the software by themself and thank to vgate for this. If they do not know how to compile the app by themself, they should ask vgate how to compile the app. This would add support burden to vgate and i think this would be a good result. Vgate people seem to know how to compile AndrOBD and add crap to it, so they can from now on explain this to all vgate customers that want to use AndrOBD ;)

aha999, 2024-07-27:
I bought two Vgate adapters since they're cheap and AndrOBD compatible. I think AndrOBD should not punish its users, since this makes AndrOBD worse, not better and the users of AndrOBD would suffer and subsequently not use the app anymore...

I think this incident should have a social media post about how unauthorized Vgate cloned an app, like "beware of the cloned app from the Vgate etc." And how we invite Vgate to be better and include AndrOBD in the list of the supported software. Maybe Vgate would change their behaviour then, especially when users would know about it.

One partial solution so manufacturers could include AndrOBD to the list of supported software would be, that AndrOBD would include itself in the Google Play store, and also gain more users/contributors. Currently manufacturers are probably not going to promote AndrOBD since Google Play is sadly the only official Android app store.