fr3ts0n/AndrOBD #210: The application drains power when OBD not connected

jbrepogmailcom, 2022-07-24:
# Describe the bug

When Bluetooth OBD sensor is turned off, the application is still actively running and is draining battery, in my case 60% of battery per day.

I recommend that when OBD is disconnected and the app is not in foreground, it should go to deep sleep for 30 sec. and then check for connection again. If not connect, re-enter deep sleep.

## To Reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior

A clear and concise description of what you expected to happen.

## AndrOBD Debug log files

Please attach AndrOBD debug log files.

AndrOBD stores log files in a subdirectory on your primary storage device. (com.fr3ts0n.ecu.gui.androbd/log/AndrOBD.log.*)

### Log detail level

The detail level of the logged debug information can be adjusted in the Settings > Development options > Logging level.

By default the detail level is set to INFO.
Preferred detail level for issue analysis is DEBUG. (Since V1.5.0 this level is called FINE)

## Important details on your environment:

- Android OS version:
- AndrOBD version:
- OBD adapter type:
- Connected vehicle make, type, year, engine size:

## Additional context

Add any other context about the problem here.

fr3ts0n, 2022-07-29:
Thank you for your report.
AndrOBD does NOT automatically re-try a stopped/broken bluetooth connection. (see also comment on #213 )

However, the app is started with the "Keep awake" flag at startup which currently stays active even when connection stops.
I'll have a look at this behaviour.

jbrepogmailcom, 2022-07-29:
I have automation to turn off AndrOBD when not in use - see case #210

jbrepogmailcom, 2022-08-04:
Yes please, that is serious problem for me. I would like to automate as much as possible and have the mobile run unattended. The problem is that even when I close the application via UI interactions, the process in background still runs and it keeps wakelock, so battery drains very quickly. I had to create another UI interaction that kills the app through app management is settings, but that is VERY complex.

May I recommend that application responds to some automation intents?
- disconnect
- connect
- display data screen (for some reason the mqtt plugin does not send data until the data screen is displayed..)
- close background process for bot main app and the mqtt plugin

This would make your application the killer one! :-)

fr3ts0n, 2022-08-05:
For unattended use the "Keep awake" config can be switched OFF  in user settings:
Common settings > Keep main screen on

Automated connection to previous ELM adapter, vehicle and display data screen with last items selected, may all be automated at APP startup in user settings:
Common settings > Use last settings
(see also here: [Wiki](https://github.com/fr3ts0n/AndrOBD/wiki/Automation-on-startup) )

Any intent of connect would need to assume that the last, remembered  adapter / vehicle is re-connected. This automation is already handled with app startup (as described above)

With these above configs set, all you need to automate is to start- and stop the app.

However, when AndrOBD is closed by UI, then also the communication thread is stopped, as well as the connected plugins, so there should NOT be any thread running any more.
- Please could you provide a LOG file (FINE) to verify this?

Thx

jbrepogmailcom, 2022-08-05:
Hello, thanks very much, I appeciate the work you do, really! And I think the program is great and best in the area that can be. I would like to make a correction to "For unattended use the "Keep awake" config can be switched OFF in user settings: Common settings > Keep main screen on"

I have this setting off and after I quit the application by pressing BACK three times and turned sceen off, the phone never entered deep sleep, because of _some_ wakelock. After I went to application settings and force-killed the AndrOBD process manually, the wakelock was gone and phone is now 95% in deep sleep. So I think that your statement above is not fully correct, because even with screen off, the CPU of the phone is kept running by the background process.

Also other statements are not correct in my case, it may be caused by the phone. When I start AndrOBD it does not connect automatically and does not show data screen automatically, even if I have all "remember" settings on. I have to manually press the connection button and manually select data screen. Always after start and after OBD device was off.

Jan

fr3ts0n, 2022-08-06:
Please could you provide a log file (FINE) to get an idea of what happens during startup and shutdown sequence?
Thx