fr3ts0n/AndrOBD #211: What ODB communication protocol to use?

jbrepogmailcom, 2022-07-24:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

When I connect ELM327 to canbus in my Jeep 2016, strange things happen. I get different errors reported by car, MIL turns on, Today even the power steering stopped working.
I woul assume that ELM327 is readonly, but it apparently breaks CANBUS communication. Is there any advice how to correct it? I can set various communication protocols or timing, maybe that can help?

I was googling what communication protocols various manufacturers use, but could not find any. Maybe author of this application knows? Thanks

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

fr3ts0n, 2022-07-29:
The most typical, current CAN protocol is *ISO 15765-4 CAN (11 Bit ID, 500 KBit)*

This is also part of ELM's automatic protocol detection algorithm, so it usually is NOT necessary to be selected manually.

jbrepogmailcom, 2022-07-29:
Automatic detection worked, but found different protocol, that caused many failures to be reported in my car and even turned power steering off - potentially dangerous situation!

After I set the protocol manually to one I found in forums - and the one you mentioned, problems were gone. I would suggest to start automatic detection with exactly that protocol.

I used torque that have automatic detection and it was causing troubles with car as well

fr3ts0n, 2022-07-29:
The sequence of automatic protocol detection is pre-defined in the adapter's firmware. 
The app can not influence this sequence ... It can either select the protocol manually, or rely on pre-defined detection sequence.

closed 2022-07-29 by jbrepogmailcom