fr3ts0n/AndrOBD #212: Does not reconnect after bluetooth adapter off/on - does not run unattended

jbrepogmailcom, 2022-07-25:
# Describe the bug

When I turn off the car the ELM327 turns off. After next start, adapter turns ON but the AndrOBD does notreceive any data. It says connected, but "No data". I have to manually disconnect and reconnect again.

If there is intent for connection, I could automate it with MacroDroid so it automatically reconnects after ELM327 detected. Is there such intent?

## To Reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior

A clear and concise description of what you expected to happen.

## AndrOBD Debug log files

Please attach AndrOBD debug log files.

AndrOBD stores log files in a subdirectory on your primary storage device. (com.fr3ts0n.ecu.gui.androbd/log/AndrOBD.log.*)

### Log detail level

The detail level of the logged debug information can be adjusted in the Settings > Development options > Logging level.

By default the detail level is set to INFO.
Preferred detail level for issue analysis is DEBUG. (Since V1.5.0 this level is called FINE)

## Important details on your environment:

- Android OS version:
- AndrOBD version:
- OBD adapter type:
- Connected vehicle make, type, year, engine size:

## Additional context

Add any other context about the problem here.

fr3ts0n, 2022-07-29:
AndrOBD does re-try to connect on the vehicle side if the vehicle doesn't respond, but the adapter is still connected.
However, after the adapter gets disconnected there is NO re-try. This represents the common situation of leaving the car/bluetooth range with the mobile.

There is a intent for the connection attempt, but this is currently only internal, but NOT exported to the system.

jbrepogmailcom, 2022-07-29:
As in other case, I have to detect that car is moving (by macrodroid ACTIVITY trigger) and the use UI interaction (simulate taps) to connect again.

wunter8, 2023-06-21:
I have also wanted a way to automatically connect everything once I get back into the car, so I can automatically start sending MQTT data using the plugin.

I might use an automation app to simulate taps, too, but it seems like exposing the internal intent would be very convenient. How hard would it be to expose that externally?

jkaminsk1979, 2026-09-05:
Maybe I am too late to join this topic, but curently testing with real adapter on my motorbike with laptop and phone in hands so can see real logs and cat log.
WriteTelegram  in StreamHandler even catch exeption, never send it into higher level so StreamHandler.run() never breaks. 
public int writeTelegram(final char[] buffer, int type, Object id)
	{
		int result = buffer.length;
		
		// Send data threaded to de-couple from main thread
		new Thread(new Runnable()
		{
			@Override
			public void run()
			{
				try
				{
					String msg = new String(buffer);
					msg += "\r";
					
					log.finer(this.toString() + " TX:"
					          + ProtUtils.hexDumpBuffer(msg.toCharArray()));
					
					out.write(msg.toCharArray());
					out.flush();
				}
				catch (Exception ex)
				{
					log.severe("TX error:'"
					           + ProtUtils.hexDumpBuffer(buffer) + "':"
					           + ex.getStackTrace());
				}
			}
		}).start();
		return (result);
	}
===
my logs shows this with broken pipe. :
026-09-05 11:09:21.015  3502-19275 StreamHandler           com.wzh.yuvwater.demo                D  ✅ TX successful: 'ATPC'
2026-09-05 11:09:21.015  3502-19274 StreamHandler           com.wzh.yuvwater.demo                E  ❌ TX error: 'ATPC' | Broken pipe (Fix with AI)
                                                                                                    java.io.IOException: Broken pipe
                                                                                                    	at android.net.LocalSocketImpl.writeba_native(Native Method)
                                                                                                    	at android.net.LocalSocketImpl.-$$Nest$mwriteba_native(Unknown Source:0)
                                                                                                    	at android.net.LocalSocketImpl$SocketOutputStream.write(LocalSocketImpl.java:144)
                                                                                                    	at android.bluetooth.BluetoothSocket.write(BluetoothSocket.java:673)
                                                                                                    	at android.bluetooth.BluetoothOutputStream.write(BluetoothOutputStream.java:79)
                                                                                                    	at sun.nio.cs.StreamEncoder.writeBytes(StreamEncoder.java:221)
                                                                                                    	at sun.nio.cs.StreamEncoder.implWrite(StreamEncoder.java:282)
                                                                                                    	at sun.nio.cs.StreamEncoder.write(StreamEncoder.java:125)
                                                                                                    	at java.io.OutputStreamWriter.write(OutputStreamWriter.java:207)
                                                                                                    	at java.io.BufferedWriter.write(BufferedWriter.java:184)
                                                                                                    	at java.io.Writer.write(Writer.java:213)
                                                                                                    	at com.chigee.pm.server.obd.StreamHandler$1.run(StreamHandler.java:70)
                                                                                                    	at java.lang.Thread.run(Thread.java:1012)
2026-09-05 11:09:21.016  3502-17947 StreamHandler           com.wzh.yuvwater.demo                V  📥 processRxChar: 41 ('A')
2026-09-05 11:09:21.016  3502-17947 StreamHandler           com.wzh.yuvwater.demo 
===
It hapens when adapter is "frozen" or somehow behave incorectly. But even after broken pipe, we have response from the ELM adapter. So bt connection is alive. We can blue and green diods blinking. 
What esle is very often happaning is the communication is "frozen" for some time (CONNECTED, but we cant see data are changing, even ELM protocool says connected). Than StreamHandler.ru() is hangig forever. The only one place to do check is when we sleep(1s), we can add counter like this :

public void run() {
		int chr;
		Log.d(TAG, "🟢 RX Thread started");
		log.info("RX Thread started");
		waitingPing = false;

		try {
			while (true) {
				// KONTROLA – zda InputStream existuje
				if (in == null) {
					Log.e(TAG, "❌ InputStream is NULL!");
					break;
				}
[... 28 lines cut ...]
							Log.w(TAG, "⚠️ No data for 5s - sending ping (AT)");
							log.warning("⚠️ No data for 5s – sending AT ping...");
							writeTelegram("AT\r".toCharArray());

							//log.warning("⚠️ No data for 5s – sending ATZ...");
							//writeTelegram("ATZ\r".toCharArray());

							//CommService.elm.reset();
						}

					}
				}
			}
		} catch (Exception ex) {
			Log.e(TAG, "❌ RX error: " + ex.getMessage(), ex);
			log.log(Level.WARNING, "RX error", ex);
		}

		Log.d(TAG, "🔴 RX Thread stopped");
		log.info("RX Thread stopped");
	}

===
so we can decide, if we will wait for 5s periods forever, or uncomment the break and will finaly break StreamHandler a call connectionlost() in BtCommService. 
I am stil doing more tests with various adapters in different situation scenario. I relay hope owner will look at this and say something.