fr3ts0n/AndrOBD #213: Car battery drained

jbrepogmailcom, 2022-07-25:
# Describe the bug

This happened several times, this time I was lucky to notice soon. When the app stays connected over ELM327, sometimes the car can power itself off properly. But once, the car battery was completely depleted next day Morning and Today I noticed, that when AndrOBD is running and connected, the car tries to power off (few minutes after switching ignition off), but imediately turns on again. Seems to me like some problem with ELM327 communication on CANBUS, but only when AndrOBD is connected. I use protocol ISO 15765-4 CAN 11 bit ID, 500 KBit). When I disconnected the ELM everything was fine.

Can the problem be really related to AndrOBD and did it happen to anyone else?

## To Reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior

A clear and concise description of what you expected to happen.

## AndrOBD Debug log files

Please attach AndrOBD debug log files.

AndrOBD stores log files in a subdirectory on your primary storage device. (com.fr3ts0n.ecu.gui.androbd/log/AndrOBD.log.*)

### Log detail level

The detail level of the logged debug information can be adjusted in the Settings > Development options > Logging level.

By default the detail level is set to INFO.
Preferred detail level for issue analysis is DEBUG. (Since V1.5.0 this level is called FINE)

## Important details on your environment:

- Android OS version:
- AndrOBD version:
- OBD adapter type:
- Connected vehicle make, type, year, engine size:

## Additional context

Add any other context about the problem here.

fr3ts0n, 2022-07-29:
I never thought about that AndrOBD app would stay running with connected adapter in the vehicle.

It is definitely possible that the connected app re-tries to connect to the vehicle, and these re-tries cause the vehicle to keep the ECU's and CAN network alive. In consequence this results in high battery drain.

Please could you explain the use case for that?

jbrepogmailcom, 2022-07-29:
Use case is simple - leave the app running, turn off the car. After few minutes, the car tries to power down, but goes on again. Until the battery is depleted. Since the app does not have intent to be controlled with, I actually have a macrodroid macro, which detects that car is standing still and then turns off the AndrOBD **through UI interaction** (by simulation taps)