fr3ts0n/AndrOBD #214: Unofficial AndrOBD in the Play Store?

drws, 2022-07-28:
# Describe the bug

@fr3ts0n only mentions the [F-Droid](https://f-droid.org/) availability and even explicitly stated that in #84.

But check [this](https://play.google.com/store/apps/details?id=com.github.androbd) out. It seems there is an unofficial build distributed in the Play Store since this February.

I've compared the Play Store APK with the latest original and it is rebuilt, with changed `classes.dex` and `resources.arsc`. Exodus [returns](https://reports.exodus-privacy.eu.org/en/reports/274581/) the same as with the original: 0 trackers, 8 permissions (the same ones). This is no guarantee, just as an indication. Even if the Play Store unofficial build is currently clean, the author could add malicious code anytime.

Is there anything that needs to or can be done about it? Will there be any effect if such apps are reported to the Play Store (based on similar cases)?

fr3ts0n, 2022-07-28:
Hi @drws ,
Thank you for the information. It looks like it is a plain re-build, but I'll check on that.

I would have preferred to NOT have AndrOBD on playstore, because this also triggers 1st level support requests of some users with rather limited knowledge / interest of OBD diagnostics / adapters.
It also triggers potential signature problems if you mix up app- and -plugins between F-Droid and PlayStore.

As long as it it just a plain re-build, there is nothing about it. 
- I don't intend to spend additional time for maintaining this unofficial release
- That is also what I'll communicate towards @cppox

If there are some changes/manipulations contained, then @cppox will have to ... :
- to provide public source code
- include all additional copyright notes
... to meet the GPL license

In such a case of license violations like [vgate](https://play.google.com/store/apps/details?id=com.minisatip.vgate) I'll try to claim a license infringement and make Google remove it from PlayStore, but this always takes a lot of time, effort and nerves ... 😢 (I still didn't succeed with vgate yet, which definitely violates the license ...)

lefterisgar, 2023-01-18:
The app on Play Store has ads. A lot of them. It's a shame because AndrOBD is such a good FOSS app and @cppox takes advantage of it in order to make money.

ieffuie4, 2024-07-05:
This app is still up. And its outdated. The person who added crap to it does not even want to do basic updating work. Now its making bad reputation to AndrOBD because it have all the issues that was fixed since many months still inside.
Could some more people report it to the google service and add 1-star reviews to the app describing the problems? You seem to need a google account to be able to report and rate this bad clone.

fr3ts0n, 2024-07-06:
Yes, I have seen that. 
I have filed a license infringement claim at google to have them remove it from play store, but they don't seem to care.

ieffuie4, 2024-07-06:
Maybe file again a license infringement? In such big companies like google it depends on the person who gets the license infringement on his desk. Seem that a different one removed the vgate tool. Maybe you can link the already removed vgate app and show like this to a new google cutomer person that "again" your free software license is not respected.

aha999, 2024-07-07:
The copycat AndrOBD app is still on Google play, 1000+ downloads, making them ad-enabled money:
 https://play.google.com/store/apps/details?id=com.github.androbd