fr3ts0n/AndrOBD #215: Application freezes and becomes unresponsive often and needs to be manually killed

jbrepogmailcom, 2022-07-30:
# Describe the bug

Very often, probably in half of the cases, when tapping CONNECT, the application confirms it has been connected to adapter, then to ECU, then data screen shows, but it is completely blank. The application is frozen and does not react to taps. After a while an OS message pops out that application does not respond and can be killed by that OS dialog.

This happened almost all the time, after I set up logging to off, it seems to happen less often but still often. On subsequent start, the app connects and displays data, or freezes again, it is about 50/50.

I suspect that application freezes when it starts receiving ECU data from adapter. I tried to change all types of settings, timing, NRZ restart, not help, still freezing.

The frozen app drains battery a lot, the phone becomes hot. So the frozen communication is suspected.

Any recommendations for debugging? I could expeeriment with AT commands, but there is too many of them.

## To Reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior

A clear and concise description of what you expected to happen.

## AndrOBD Debug log files

Please attach AndrOBD debug log files.

AndrOBD stores log files in a subdirectory on your primary storage device. (com.fr3ts0n.ecu.gui.androbd/log/AndrOBD.log.*)

### Log detail level

The detail level of the logged debug information can be adjusted in the Settings > Development options > Logging level.

By default the detail level is set to INFO.
Preferred detail level for issue analysis is DEBUG. (Since V1.5.0 this level is called FINE)

## Important details on your environment:

- Android OS version:
- AndrOBD version:
- OBD adapter type:
- Connected vehicle make, type, year, engine size:

## Additional context

Add any other context about the problem here.

fr3ts0n, 2022-07-31:
Please could you provide a log file to get a glance of what happens in these situations.

I remember some adapters did cause BT problems when these were initialized with the ATZ command. 
It might be worth a try to disable that command (in developer settings) ...

jbrepogmailcom, 2022-07-31:
Sorry for stupid question, where do I find the log?

fr3ts0n, 2022-07-31:
Please see here: https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues#debug-log-files

jbrepogmailcom, 2022-08-01:
Hello I think I am not able to provide the log. When the application freezes, it looks like the log file is open for writing and after app crashes, the file has zero length. However, I will add some more details.

When I launch the application without data screen, it says "Connected", but in mqtt broker, I see that no messages are sent. Therefore data collection does not work. Car ID screen does work, but it may just display data from adapter ID, or from the past?

The app freezes exactly in the moment, when I try to display data screen. It freezes so soon that even the menu (to select data screen, car id etc...) does not completely disappear from display. It stays semi-transparent and the app is frozen and must be killed manually.

In developer settings, I disabled all AT commands, but it did not help. The app is still crashing. But not always - in about half of the cases the app runs as expected.

fr3ts0n, 2022-08-01:
Did you possibly revoke the FILE read/write permission of the app?
- This could cause an error / exception when it tries to log messages

As alternative could you try to create a log output with *adb* tool? 
```adb logcat```

jbrepogmailcom, 2022-08-01:
No the storage permission is granted. In the folder I have may log files with zero length

closed 2023-12-20 by fr3ts0n