fr3ts0n/AndrOBD #216: Add Translating section to Wiki

aha999, 2022-09-10:
**Is your feature request related to a problem? Please describe.**
There is currently no instructions section in the AndrOBD Wiki as to how to translate the app.

**Describe the solution you'd like**
Surely Weblate is not the only way of translating AndrOBD, and if it is not, instructions on translating should be made in the Wiki, on how to translate in the Weblate, and also if there are other methods of translating.

**Additional context**
If this issue gets the Translating section in the Wiki, i will update the readme and include the Translating link.

fr3ts0n, 2022-09-11:
added here:
https://github.com/fr3ts0n/AndrOBD/wiki/Language-translation

aha999, 2022-09-11:
That's awesome, so fast and well done. you amaze me...

closed 2022-09-11 by aha999